petrobras/ross #419: Improving the Tutorial - Visualizing the Rotor

rennergcm, 2020-01-17:
a. The plot shown when matplotlib is chosen is too small and hard to visualize. When "figsize" [rotor2.plot_rotor(plot_type='matplotlib',figsize=(15,15))] was added the following 
TypeError was displayed "_plot_rotor_matplotlib() got an unexpected keyword argument 'figsize'". Therefore, my advice is to add in the tutorial a code in which the user can change the figure_size.

b. It code described below ran using bokeh version 1.4 only. Therefore, users should be warned which version of bokeh should be installed.

c. The code "rotor1.plot_rotor(plot_type='bokeh')" should be replaced by 'show(rotor1.plot_rotor(plot_type='bokeh'))'

raphaeltimbo, 2020-02-05:
To have a matplotlib of a different size you can use the fact that an axes can be passed as an argument to the plot function:
```python
# create the figure with the desired size
fig, ax = plt.subplots(figsize=(15, 15))
# pass the created axes so that the plot function can draw on it
ax = rotor.plot_rotor(plot_type='matplotlib', ax=ax)
```
Notice that the ax is also returned by the `rotor.plot_rotor()` method and you can change it as you want.
We are planning to have the option to pass `**kwargs` to some plot functions, so the user can for example choose the color of a line when calling the plot, but this is not viable to the rotor plot, since we are drawing different objects (disks, bearings etc.).

closed 2020-02-05 by raphaeltimbo

rodrigomoliveira1, 2020-02-05:
Just to complement, you can also do it by:

```
rotor1.plot_rotor(plot_type='matplotlib')

# first, get the current figure
fig = plt.gcf()
# then, set the new size (width, height)
fig.set_size_inches(width, height)
```