petrobras/ross #426: Improving Tutorial - Mode Shapes

rennergcm, 2020-01-17:
a. How to plot this graph in 2D? I do not think mode shape in 3D useful.

speed = 10
modes = rotor1.run_modal(speed)
modes.plot_mode(0)

rodrigomoliveira1, 2020-02-19:
I did a 2D mode shape graph for api_report.py. 
I could copy / move the code to results.py for a plot2D method

raphaeltimbo, 2020-02-19:
> 
> 
> I did a 2D mode shape graph for api_report.py.
> I could copy / move the code to results.py for a plot2D method

That would be good.

rodrigomoliveira1, 2020-02-19:
Is it ok this way? Or should I add some marks to indicate the bearings locations.
I'm plotting with matplotlib. For each whirl direction, there's a different color:

- Forward: blue
- Mixed: black
- Backward: red

[image]
[image]

closed 2020-02-19 by raphaeltimbo