petrobras/ross #431: New package - Stochastic ROSS

rodrigomoliveira1, 2020-01-30:
I am opening this issue to discuss the implementation of a stochastic module for ROSS.
 
- How to choose what parameters are passed as stochastic by the user;

- Defining distributions to these parameters:
    - inputing support, 1st, 2nd, n-th momentum;
    - inputing a list with a "ready-to-go" distribution for each parameter;
    - and/or another way

- Defining the charts, graphs and maps as output;

- Defining how to adapt the API Report to the stochastic analysis;

- Other guide lines...

LuanDaSilva, 2020-02-08:
Stochastic model would recieve a numpy array generated by numpy.random."distribution".Considering this as a entry, ROSS would only identify which variable will be used to the stochastic procedure. The outputs should be the percentiles 5%,50% and 95% for FRFs and some thermal graph with dashed doted lines for Campbell diagrams

LuanDaSilva, 2020-02-08:
API tools can be the probability of not attend some rule. The entry can be some confidence interval required and the desired analysis

LuanDaSilva, 2020-02-08:
Alternatively to numpy RNG distributions, we could also create a distribution wizard, which allow the user to use the available information to obtain the most suitable distribution for each case.

LuanDaSilva, 2020-02-08:
This wizard accepts any kind of entry. However, all this freedom could be kin da unfriendly at the first glance. An alternative is let the user choose the kind of entry. "All" could use the wizard without restrictions,"Simple" only recieves mean, standart deviation and support and "Data" generates k moments from some entry data

LuanDaSilva, 2020-02-08:
In order to generate good estimatives of standard errors and confidence intervals when data is available, we can implement bootstrap methods

LuanDaSilva, 2020-02-08:
This techniques that I cited are described in Larry Wasserman's All of Statistics. Bootstrap is in Ch.8 and Moments method are in Ch.9

LuanDaSilva, 2020-02-20:
Today, 20/02/2020, I developed the code for Stochastic ross. The implementation is very redable and easy to use:
`import itertools
from collections import Iterable
def stochastic_ross(f,*args):
    f_list = []   
    mix = []
    for arg in args:
        if isinstance(arg,Iterable):
            mix.append(list(arg))
        else:
            mix.append([arg])
    new_args = list(itertools.product(*mix))
    for arg in new_args:
        f_list.append(f(*arg))
    return f_list`

rodrigomoliveira1, 2020-03-16:
Sine the PR #452, @LuanDaSilva and I have worked on how the user would instantiate functions to generate random rotors.

Assuming the user is used to the ROSS syntax, we have implemented functions that are as similar as possible to the ROSS functions so that the learning curve of the new module is smoother. Let's check some examples:

### Creating shaft elements:

#### Single shaft element (current on ROSS):
To create a very basic shaft element, the user must input, at least, a length, an inner diameter, an outer diameter and a material:

```
>>> from ross.shaft_elements import ShaftElement
>>> from ross.material import steel

>>> ShaftElement(L=0.1, idl=0, odl=0.05, material=steel)
ShaftElement(L=0.1, idl=0.0, idr=0.0, odl=0.05,  odr=0.05, material='Steel', n=None)
``` 
And a rotor shaft is created from a list of shaft elements:

```
>>> i_d = 0
>>> o_d = 0.05
>>> n = 6
>>> L = [0.25 for _ in range(n)]

>>> shaft_elem = [
...     ShaftElement(
...            l,
...            i_d,
...            o_d,
...            material=steel,
...            shear_effects=True,
...            rotary_inertia=True,
...            gyroscopic=True,
...     )
...     for l in L
... ]
>>> shaft_elem 
[ShaftElement(L=0.25, idl=0.0, idr=0.0, odl=0.05,  odr=0.05, material='Steel', n=None),
 ShaftElement(L=0.25, idl=0.0, idr=0.0, odl=0.05,  odr=0.05, material='Steel', n=None),
 ShaftElement(L=0.25, idl=0.0, idr=0.0, odl=0.05,  odr=0.05, material='Steel', n=None),
 ShaftElement(L=0.25, idl=0.0, idr=0.0, odl=0.05,  odr=0.05, material='Steel', n=None),
 ShaftElement(L=0.25, idl=0.0, idr=0.0, odl=0.05,  odr=0.05, material='Steel', n=None),
 ShaftElement(L=0.25, idl=0.0, idr=0.0, odl=0.05,  odr=0.05, material='Steel', n=None)]
```
So, the objective here is to keep most of this structure when planning the stochastic ross algorithm.

#### Random shaft element (current on ROSS):
To create a very basic radom shaft element, the user must input, at least, a length, an inner diameter, an outer diameter and a material, as shown above.
However, there're two main differences here are:
- the user must indicate what arguments are passed as random;
    - e.g.: `["L"]` or `["L", "odl"]`, or `["odl", "odr", "material"]`, etc...
- the output is a list of `n` shaft elements
    - it doesn't mean a shaft is being created, it means that are `n` possibilities of shaft elements to fit a specific position in a given rotor shaft.

So, to keep the standards, there's a class to create these random elements: `ST_ShaftElement` (all the methods and classes related to stochastic ross receives a `ST_` as prefix.

```
class ST_ShaftElement():
    """Random shaft element

    Creates an object containing a list with random instances of ShaftElement

    Parameters
    ----------
    L : float, pint.Quantity, list
        Element length.
        Input a list to make it random.
    idl : float, pint.Quantity, list
        Inner diameter of the element at the left position.
        Input a list to make it random.
[... 67 lines cut ...]
    ...                            idr=0,
    ...                            odr=np.random.uniform(0.1, 0.2, 5),
    ...                            material=steel,
    ...                            is_random=["odl", "odr"],
    ...                            iteration=False,
    ...                            )
    >>> len(elements.elm_list)
    5
```

And a random rotor shaft can be created just like the first example, but adding some of ST_ShaftElement object when the user wants an element to be random.

```
>>> shaft_elem = [
...    ShaftElement,
...    ShaftElement,
...    ST_ShaftElement,
...    ST_ShaftElement,
...    ShaftElement,
...    ShaftElement,
...    ShaftElement,
... ]
```

This logic shall apply to every other element while instantiating a random rotor object.

closed 2020-04-13 by rodrigomoliveira1