petrobras/ross #442: fluid_flow tests are taking too long to run

raphaeltimbo, 2020-02-12:
The tests on fluid flow are taking a long time to run. Can we simplify the tests so they can faster?
@JuliaMota Can you look at this?

JuliaMota, 2020-02-12:
Ok!  I will work on it.

JuliaMota, 2020-02-16:
The changes made by pull request #446 reduced the test time from 7m to 3m, approximately. The longest delay occurs in the function 'test_find_equilibrium_position' because it is an iterative method. 
Is the time spent good for testing or is it necessary to reduce more?

raphaeltimbo, 2020-02-17:
Is it possible to reduce the `test_find_equilibrium_position` by reducing the number of iterations (or maybe reducing the tolerance)? If you could reduce more time in this test it would be great, but if it is not possible I think the reduction you've already achieved is good enough and we can merge the PR.

JuliaMota, 2020-02-18:
I made another modification reducing the number of iterations of the function `test_find_equilibrium_position`.
Initially the fluid-flow tests took 7m 20s. After the initial modifications this time was reduced to 32s. Now, on my computer, it's taking about 15s.

Is this a good time? If the answer is affirmative, this issue can be closed after merging the PR.

raphaeltimbo, 2020-02-18:
> I made another modification reducing the number of iterations of the function `test_find_equilibrium_position`.
> Initially the fluid-flow tests took 7m 20s. After the initial modifications this time [...]

Yes. This is good. I've already merged #446 
Thanks!

closed 2020-02-18 by raphaeltimbo