petrobras/ross #445: Material accepts inconsistent arguments

raphaeltimbo, 2020-02-14:
Material accepts inconsistent arguments during instantiation. See example below.

```python
>>> steel = rs.Material(name="Steel", rho=7810, E=211e9, G_s=81.2e9)
>>> steel.Poisson
0.2992610837438423
>>> steel = rs.Material(name="Steel", rho=7810, E=211e9, G_s=81.2e9, Poisson=0.2)
>>> steel.Poisson
0.2
```

raphaeltimbo, 2020-02-14:
I am thinking about removing `Poisson` as an argument for material instantiation.
Having three arguments when one is just a combination of the other two can cause inconsistency in several places: during instantiation, when saving (which arguments should we save? All three?), when loading from memory (again, if we saved all three arguments like we do now, which arguments should we use to re-instantiate from memory).
My idea is to have `Poisson` only as a method to avoid these problems. Other software only use two properties to define the material (e.g. XLTRC uses E and G_s)

rodrigomoliveira1, 2020-02-14:
Or, we could block the user instantiating 3 arguments at once. Allowing the user only to choose 2 between them.

If the user tries to pass all 3 at once, the code raises an exception.

PR #453 (2020-02-27): Material units
This adds check_units to the `Material` class:

```python
import ross as rs

Q_ = rs.units.Q_
steel = rs.Material(name='steel', rho=7810, E=Q_(3e7, 'psi'), G_s=Q_(1e7, 'psi'))
```

It also closes #445
closed 2020-02-27 by raphaeltimbo