petrobras/ross #458: Binder examples error

srmnitc, 2020-03-05:
This is part of my review of the software paper at openjournals/joss-reviews#2120

While trying out the examples on binder, evaluating the first cell in the notebook `tutorial.ipynb`, the following error is generated-

```
FileNotFoundError                         Traceback (most recent call last)
/srv/conda/envs/notebook/lib/python3.7/site-packages/pint/registry.py in load_definitions(self, file, is_resource)
    523                 else:
--> 524                     with open(file, encoding="utf-8") as fp:
    525                         return self.load_definitions(fp, is_resource)

FileNotFoundError: [Errno 2] No such file or directory: '/srv/conda/envs/notebook/lib/python3.7/site-packages/ross/new_units.txt'
```

raphaeltimbo, 2020-03-05:
Hi there @srmnitc !

This error is related to a new feature that we have in the master branch but is not present in the v0.2.0 (which is the released version submitted to JOSS).

We have already noticed some issues regarding the current released version (v0.2.0) and parts of the documentation that are tracking the master branch.
Is it ok if we make a new release (v0.3.0) and update the JOSS issue information below?

 
[image]

srmnitc, 2020-03-05:
@raphaeltimbo Thanks for the quick reply! From what I have seen, I think its fine to change the version. The journal would provide you an option in the end to change version number. I was able to do this for my own publication. I think getting the examples working on binder is a great step! Thanks!

raphaeltimbo, 2020-03-05:
Great! We will release a new version and fix the examples on binder as soon as possible.

raphaeltimbo, 2020-03-06:
@srmnitc , we have just released the v0.3.0.
We have also updated our docs and our binder link to track the 0.3 branch.
I have made some tests and things seem to be working. Let us know if you have any problems.
Thanks!

srmnitc, 2020-03-06:
@raphaeltimbo thanks for the quick update. I will check again. I will keep this issue open for now so that if further issues come up I will log it here.

closed 2020-04-13 by raphaeltimbo