petrobras/ross #461: Use a standard/common approach to declaring dependencies

nnadeau, 2020-03-06:
Currently, the dependencies are defined in `setup.py`:

```python
REQUIRED = [
    "numpy",
    "scipy",
    "matplotlib",
    "toml",
    "pandas>=0.23",
    "bokeh",
    "coverage",
    "xlrd",
    "pint",
]
```

In the Python ecosystem, dependencies are typically defined in a `requirements.txt` file or using a dependency management framework (e.g., poetry, pipenv).

It would be preferable that repo be formatted such that there is a separation of concerns between the `setup.py` and the declaration of dependencies. 

If this repo is intended as a library (i.e., integrated into other packages), then a `requirements.txt` with minimum versions declared is preferred. 

If this repo is intended as an end-use application (i.e., not integrated into other packages), then using `poetry` or `pipenv` is preferred.

cc https://github.com/openjournals/joss-reviews/issues/2120

raphaeltimbo, 2020-03-06:
The repo is intended as a library. We actually had a `requirements.txt` file but we have deleted because we were afraid of having duplicated information that could go out of sync.
But I undestand how not having the `requirements.txt` could be a problem in some cases. 
Doing some research I have found the following in [this blog post](https://caremad.io/posts/2013/07/setup-vs-requirement/):

>The “Library” and “Application” distinction is all well and good, but whenever you’re developing a Library, in a way it becomes your application. You want a specific set of dependencies that you want to fetch from a specific location and you know that you should have abstract dependencies in your setup.py and concrete dependencies in your requirements.txt but you don’t want to need to maintain two separate lists which will inevitably go out of sync. As it turns out pip requirements file have a construct to handle just such a case. Given a directory with a setup.py inside of it you can write a requirements file that looks like:
```
--index-url https://pypi.python.org/simple/
-e .
```
> Now your pip install -r requirements.txt will work just as before. It will first install the library located at the file path . and then move on to its abstract dependencies, combining them with its --index-url option and turning them into concrete dependencies and installing them.

Do you think this could be a solution in our case?

nnadeau, 2020-03-22:
@raphaeltimbo 

I personally recommend one of the following:

1) A super packaging framework https://python-poetry.org/

2) Reading your `requirements.txt` file into your `setup.py`:

```
with open('requirements.txt') as f:
    requirements = f.read()

setup(
	requires=requirements
)
```

nnadeau, 2020-03-22:
e.g., https://www.reddit.com/r/Python/comments/3uzl2a/setuppy_requirementstxt_or_a_combination/cxk3z76?utm_source=share&utm_medium=web2x

PR #500 (2020-03-26): Add requirements.txt file
This PR adds the requirements.txt (closes #461) file and also solves the issue regarding the new_units.txt file missing after installation.
closed 2020-03-26 by rodrigomoliveira1