petrobras/ross #463: Lint codebase using flake8

nnadeau, 2020-03-06:
It's important in OSS projects to have the codebase conform to a style guide. In Python, `flake8` is one of the de facto standard. The following issues should be fixed and `flake8` should be added to the testing suite.

```
ross/results.py:8:80: E501 line too long (88 > 79 characters)
ross/results.py:9:80: E501 line too long (85 > 79 characters)
ross/results.py:69:80: E501 line too long (80 > 79 characters)
ross/results.py:73:80: E501 line too long (97 > 79 characters)
ross/results.py:182:40: E203 whitespace before ':'
ross/results.py:315:80: E501 line too long (81 > 79 characters)
ross/results.py:456:13: F401 'mpl_toolkits.mplot3d.Axes3D' imported but unused
ross/results.py:463:80: E501 line too long (83 > 79 characters)
ross/results.py:497:16: W605 invalid escape sequence '\o'
ross/results.py:499:21: W605 invalid escape sequence '\_'
ross/results.py:701:80: E501 line too long (81 > 79 characters)
ross/results.py:709:80: E501 line too long (86 > 79 characters)
ross/results.py:712:80: E501 line too long (87 > 79 characters)
ross/results.py:715:80: E501 line too long (84 > 79 characters)
ross/results.py:718:80: E501 line too long (83 > 79 characters)
ross/results.py:773:80: E501 line too long (86 > 79 characters)
ross/results.py:836:80: E501 line too long (87 > 79 characters)
ross/results.py:895:80: E501 line too long (84 > 79 characters)
ross/results.py:896:80: E501 line too long (83 > 79 characters)
ross/results.py:897:80: E501 line too long (86 > 79 characters)
ross/results.py:988:80: E501 line too long (88 > 79 characters)
ross/results.py:1026:32: W605 invalid escape sequence '\o'
ross/results.py:1061:41: W605 invalid escape sequence '\m'
ross/results.py:1168:80: E501 line too long (80 > 79 characters)
ross/results.py:1407:40: W605 invalid escape sequence '\m'
ross/results.py:1576:35: W291 trailing whitespace
ross/results.py:1700:80: E501 line too long (93 > 79 characters)
ross/results.py:1743:80: E501 line too long (87 > 79 characters)
ross/results.py:1922:37: E131 continuation line unaligned for hanging indent
ross/results.py:2071:32: E203 whitespace before ':'
ross/results.py:2071:47: E203 whitespace before ':'
ross/results.py:2078:80: E501 line too long (80 > 79 characters)
ross/results.py:2146:80: E501 line too long (81 > 79 characters)
ross/results.py:2426:80: E501 line too long (84 > 79 characters)
ross/results.py:2427:80: E501 line too long (80 > 79 characters)
ross/results.py:2447:80: E501 line too long (87 > 79 characters)
ross/results.py:2527:80: E501 line too long (86 > 79 characters)
ross/results.py:2574:80: E501 line too long (84 > 79 characters)
ross/results.py:2668:13: F401 'mpl_toolkits.mplot3d.Axes3D' imported but unused
ross/api_report.py:7:1: F401 'pathlib.Path' imported but unused
ross/api_report.py:8:1: F401 'ross.rotor_assembly.rotor_example' imported but unused
ross/api_report.py:15:1: F401 'bokeh.layouts.gridplot' imported but unused
ross/api_report.py:97:80: E501 line too long (81 > 79 characters)
ross/api_report.py:108:80: E501 line too long (81 > 79 characters)
ross/api_report.py:113:80: E501 line too long (81 > 79 characters)
ross/api_report.py:116:80: E501 line too long (81 > 79 characters)
ross/api_report.py:121:80: E501 line too long (81 > 79 characters)
ross/api_report.py:124:80: E501 line too long (81 > 79 characters)
ross/api_report.py:187:80: E501 line too long (81 > 79 characters)
ross/api_report.py:329:80: E501 line too long (80 > 79 characters)
ross/api_report.py:354:80: E501 line too long (89 > 79 characters)
ross/api_report.py:363:80: E501 line too long (87 > 79 characters)
ross/api_report.py:371:80: E501 line too long (81 > 79 characters)
ross/api_report.py:475:17: F841 local variable 'SM_ref' is assigned to but never used
ross/api_report.py:699:80: E501 line too long (87 > 79 characters)
ross/api_report.py:782:80: E501 line too long (85 > 79 characters)
ross/api_report.py:822:80: E501 line too long (86 > 79 characters)
ross/api_report.py:860:80: E501 line too long (82 > 79 characters)
ross/api_report.py:894:80: E501 line too long (96 > 79 characters)
ross/api_report.py:984:80: E501 line too long (82 > 79 characters)
ross/api_report.py:1057:1: W293 blank line contains whitespace
ross/api_report.py:1081:80: E501 line too long (91 > 79 characters)
ross/api_report.py:1082:80: E501 line too long (83 > 79 characters)
ross/shaft_element.py:10:1: F401 'ross' imported but unused
ross/shaft_element.py:12:1: F401 'ross.materials.steel' imported but unused
ross/shaft_element.py:231:80: E501 line too long (87 > 79 characters)
ross/shaft_element.py:246:80: E501 line too long (82 > 79 characters)
[... 2237 lines cut ...]
ross/fluid_flow/fluid_flow_graphics.py:150:80: E501 line too long (80 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:162:80: E501 line too long (112 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:190:80: E501 line too long (80 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:207:80: E501 line too long (82 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:235:80: E501 line too long (95 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:252:80: E501 line too long (80 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:254:80: E501 line too long (81 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:264:80: E501 line too long (112 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:276:80: E501 line too long (80 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:278:80: E501 line too long (81 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:286:80: E501 line too long (93 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:324:80: E501 line too long (93 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:334:80: E501 line too long (104 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:346:80: E501 line too long (80 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:348:80: E501 line too long (89 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:358:80: E501 line too long (112 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:411:80: E501 line too long (81 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:428:80: E501 line too long (80 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:440:80: E501 line too long (112 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:465:80: E501 line too long (85 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:479:80: E501 line too long (80 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:491:80: E501 line too long (112 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:497:80: E501 line too long (83 > 79 characters)
ross/fluid_flow/fluid_flow_graphics.py:507:80: E501 line too long (83 > 79 characters)
```

rodrigomoliveira1, 2020-03-11:
@nnadeau 

We're currently using [Black](https://black.readthedocs.io/en/stable/) as code style guide. PR #468 has formated all the files and cleaned some of these warnings presented. Others would persist because Black is not as strict as flake8 (Black defaults to 88 characters per line, instead of 79). 
Besides, some of these warnings are on purpose (manly on test files), for better visualization and understanding of the users / devs. 

Would it be a okay to keep following Black as code style instead of flake8?

nnadeau, 2020-03-22:
yes, `black` is preferred for code formatting. `flake8` is best used as a python linter. 

you can copy the `.flake8` config file found here: https://github.com/psf/black/blob/master/.flake8

nnadeau, 2020-04-05:
fixed in https://github.com/ross-rotordynamics/ross/pull/496

closed 2020-04-05 by nnadeau