petrobras/ross #471: pip install does not work

srmnitc, 2020-03-10:
This is part of my review of the software paper at openjournals/joss-reviews#2120

I tried the installation as mentioned in the documentation in a clean environment with python 3.8.
Upon import the module as,
```
import ross as rs
```
the error I get is,
```
  File "~/anaconda3/envs/testross/lib/python3.8/site-packages/pint/registry.py", line 532, in load_definitions
    raise ValueError("While opening {}\n{}".format(file, msg))
ValueError: While opening ~/anaconda3/envs/testross/lib/python3.8/site-packages/ross/new_units.txt
[Errno 2] No such file or directory: '~/anaconda3/envs/testross/lib/python3.8/site-packages/ross/new_units.txt'
```

srmnitc, 2020-03-10:
I can confirm that installing from the repo using `pip install git+https://github.com/ross-rotordynamics/ross.git` works fine. Maybe the pypi version is not updated?

closed 2020-03-26 by rodrigomoliveira1

raphaeltimbo, 2020-03-26:
Hi there @srmnitc ! We have updated the `MANIFEST.in` file to add the missing file to the pypi package. I have tested and it looks like we solved the problem. Feel free to reopen the issue if you have any problems!

commit 5ccac70a (2020-03-26): Add new_units.txt to pypi package

The new_units.txt file was added to the installation, but it was not
available in the pypi distribution.

Closes #471.
Closes #497.