petrobras/ross #472: Warnings with tests

srmnitc, 2020-03-10:
This is part of my review of the software paper at openjournals/joss-reviews#2120

The tests run as described, although I do not consider the following issue as blocking acceptance, the tests raise a fair number of `DeprecationWarning`.  The first set in from `results.py` - a lot of warnings for invalid escape sequence and the rest are from Bokeh. I will it would be great and potentially would remove a lot of confusion is these are prevented.

rodrigomoliveira1, 2020-03-25:
Hello @srmnitc 

Those `DeprecationWarning`'s about invalid escape sequence happens  while using LaTeX syntax to write greek letters in matplotlib plot. And about Bokeh, they're happening due their recent updates to version 1.4.0 and 2.0.0. I'm going to fix most of them.

About the tests:
- ``test_rotor_assembly.py`: we've skipped some tests that returns matrices with very small values (e.g. 1e-18, 1e-19). As the computational precision does not exceed 16 decimal digits, each time the tests are run, these values appear different (although this does not imply wrong final results). In fact, we have added some explaination on the command line why we're skipping these tests.

- `test_units.py`: the tests we're skipping are related to future features. So as soon we implement them, we'll turn these tests on back again.

srmnitc, 2020-03-25:
Great, thanks for the update!

rodrigomoliveira1, 2020-03-30:
Hello @srmnitc,

I believe these 2 PR's should end those `DeprecationWarning`.

closed 2020-04-03 by srmnitc