petrobras/ross #475: Citations in the paper

srmnitc, 2020-03-10:
This is part of my review of the software paper at openjournals/joss-reviews#2120

About the paper-
- Timoshenko beam theory needs a citation
- Campbell diagram probably needs a citation

srmnitc, 2020-04-03:
Hi @raphaeltimbo how is this issue progressing? It would be last issue I have, once this is fixed, please let me know and I would be happy to go through once more and finalize the review.

rodrigomoliveira1, 2020-04-03:
Hi, @srmnitc. We have solved this issue with these last commits.
- Commit [740c750](https://github.com/rodrigomoliveira1/ross/commit/740c7509b314799e8f7c25bbca0ea56a5bf40859) - adds a reference for Timoshenko beam theory.
- Commits [724baa9](https://github.com/ross-rotordynamics/ross/commit/724baa9e1b9f78f050d1fc9120390ebd261504e8) and [f115983](https://github.com/rodrigomoliveira1/ross/commit/f1159831ab243c31c9b1ae805e9baa180bcdd045) - change the map's name to "whirl speed map", which is more general.

srmnitc, 2020-04-03:
Great thanks!

closed 2020-04-03 by srmnitc