petrobras/ross #484: Better explain citations

nnadeau, 2020-03-22:
re: https://github.com/ross-rotordynamics/ross/blob/master/paper/paper.md#L50

- how do these citations simulate the behavior of mechanical systems?
- what makes them special?
- why cite them?

cc https://github.com/openjournals/joss-reviews/issues/2120

raphaeltimbo, 2020-04-02:
They have a similar approach, so we have decided to remove the citations and keep only Friswell, 2010, which we have used also to improve the explanation on the modeling assumptions (#488).

nnadeau, 2020-04-05:
@raphaeltimbo please link the PR/commit that resolves this

nnadeau, 2020-04-05:
@raphaeltimbo from what I see (in https://github.com/ross-rotordynamics/ross/pull/504), all the citations are still there?

raphaeltimbo, 2020-04-05:
Hi there @nnadeau !
I have removed the citations on 3941ca78fada5882f195243de9241ecb34de1f68.

raphaeltimbo, 2020-04-05:
I have also removed the references from the bib file - 1ec02d4.

nnadeau, 2020-04-05:
fixed in https://github.com/ross-rotordynamics/ross/commit/3941ca78fada5882f195243de9241ecb34de1f68 and https://github.com/ross-rotordynamics/ross/commit/1ec02d473049bd21bce884a9f736d73113a4fdc6

closed 2020-04-05 by nnadeau