petrobras/ross #492: `software is extensible` claim

nnadeau, 2020-03-22:
`The software is extensible and new elements or new types of analysis can be easily included.` 

What kind of extensibility? Any examples?

cc https://github.com/openjournals/joss-reviews/issues/2120

raphaeltimbo, 2020-04-02:
We have added the following to the text: 

_"ROSS is extensible and new elements, such as different types of bearings or seals, can be added to the code."_

nnadeau, 2020-04-04:
@raphaeltimbo please link the PR/commit that adds this statement. Do you have examples of adding new bearings or seals?

raphaeltimbo, 2020-04-04:
Hi there @nnadeau !
I had accidentally removed the line when changing the explanation for modeling assumptions.
On #515 I have added the mentioned text and also examples on how to add new bearings.
Let me know if you have more comments.

nnadeau, 2020-04-05:
@raphaeltimbo thanks for the examples, but I wouldn't necessary add the python code to the paper (JOSS likes short papers and the code makes it verbose).

I'd recommend removing the python examples but keeping your updated statements: 

```
ROSS is extensible and new elements, such as different types of bearings or seals, can be added to the code. As anexample, one can add a class for a tapered roller bearing by inheriting from `BearingElement`:

Other elements that require more customization can be added by inheriting directly from `Element`, in this case it isnecessary to implement the required methods that should return the element's mass, stiffness, damping, and gyroscopicmatrices:
```

just reformulate the above statements to not target "in-paper examples", but rather the examples you've included in the repo

raphaeltimbo, 2020-04-05:
I have removed the Python code from the paper and added a mention to the `BallBearingElement` that we have already implemented and is available in the repo (commit 
7298d94).

raphaeltimbo, 2020-04-05:
I have modified the text to avoid repeating 'an example' ( commit 6e28f4d9a6915d355583eca4b25825b210ec3d54).
The text now is:

> ROSS is extensible and new elements, such as different types of bearings or seals, can be added
> to the code. As an example, one can add a class for a tapered roller bearing by inheriting
> from BearingElement. The implementation of the BallBearingElement in our code uses
> this strategy.
> Other elements that require more customization can be added by inheriting directly from
> Element, in this case it is necessary to implement the required methods that should return
> the element’s mass, stiffness, damping, and gyroscopic matrices.

nnadeau, 2020-04-05:
@raphaeltimbo looks good. Nitpick from me: repeated use of `as an example` two sentences in a row

```
ROSS is extensible and new elements, such as different types of bearings or seals, can be added to the code. As an 
example, one can add a class for a tapered roller bearing by inheriting from `BearingElement`. As an example we have 
the implementation of the `BallBearingElement` in the code.
```

nnadeau, 2020-04-05:
i see you caught that already in https://github.com/ross-rotordynamics/ross/commit/6e28f4d9a6915d355583eca4b25825b210ec3d54

nnadeau, 2020-04-05:
:+1: fixed in https://github.com/ross-rotordynamics/ross/commit/6e28f4d9a6915d355583eca4b25825b210ec3d54 and https://github.com/ross-rotordynamics/ross/commit/7298d94214d16526fbb3d716ee79b2dc1a8e6bc4

closed 2020-04-05 by nnadeau