petrobras/ross #493: Other works using ROSS?

nnadeau, 2020-03-22:
From https://joss.readthedocs.io/en/latest/submitting.html#what-should-my-paper-contain

```
Mention (if applicable) a representative set of past or ongoing research projects using the software and recent scholarly publications enabled by it.
```

Any citations, works, references, projects, etc. that are using ROSS? 

cc https://github.com/openjournals/joss-reviews/issues/2120

raphaeltimbo, 2020-04-02:
We have added the following to the text: 

_“ROSS has been used to evaluate the impact of damper seal coefficients uncertainties in a compressor rotordynamics [@timbo2019].  
Other projects are also using ROSS to test machine learning algorithms that can diagnose machine failure. Here we can create a model with some problem such as unbalance or misalignment and then use the output data from this model to test the learning algorithm and check the accuracy of this algorithm in diagnosing the problem.”_

nnadeau, 2020-04-04:
@raphaeltimbo please link the PR/commit that adds this statement.

raphaeltimbo, 2020-04-04:
Link to the commit: 5c0845e2b82192fa1395a0df690073b4a3bc25e1

closed 2020-04-05 by nnadeau