petrobras/ross #494: Explain significant contribution

nnadeau, 2020-03-22:
From https://joss.readthedocs.io/en/latest/submitting.html#submission-requirements

```
The software should be a significant contribution to the available open source software that either enables some new research challenges to be addressed or makes addressing research challenges significantly better (e.g., faster, easier, simpler).
```

How does ROSS fulfil the above statement?

cc https://github.com/openjournals/joss-reviews/issues/2120

raphaeltimbo, 2020-04-02:
We have modified the text and the following two parts address this JOSS requirement: 

“To our knowledge, in the rotordynamic field, ROSS is the first software being developed using the open source concept, with the code being fully available on code hosting platforms, issues tracked online, clear license and possibility of direct contribution.” 

“Developing the software using Python and its scientific ecosystem enables the user to also make use of this ecosystem, making it easier to run rotordynamics analysis. It is also easier to integrate the code into other programs, since we only use open source packages and do not depend on proprietary commercial platforms.”

nnadeau, 2020-04-04:
@raphaeltimbo please link the PR/commit that adds these statements.

Similar to https://github.com/ross-rotordynamics/ross/issues/495, I'm not convinced by:

```
“To our knowledge, in the rotordynamic field, ROSS is the first software being developed using the open source concept, with the code being fully available on code hosting platforms, issues tracked online, clear license and possibility of direct contribution.”

“Developing the software using Python and its scientific ecosystem enables the user to also make use of this ecosystem, making it easier to run rotordynamics analysis. It is also easier to integrate the code into other programs, since we only use open source packages and do not depend on proprietary commercial platforms.”
```

How is that scientifically novel? Cost is incurred whether upfront (e.g., paying for proprietary software) or in the long run (e.g., free software that is less efficient). What differentiates ROSS from `commercial finite element softwares that have a rotordynamic module` ?

raphaeltimbo, 2020-04-04:
Hi there @nnadeau !
> @raphaeltimbo please link the PR/commit that adds these statements.
> 

Link to the commit: def64b5462d227cc902cf671942101a760179270

> Similar to #495, I'm not convinced by:
>
> ```
> “To our knowledge, in the rotordynamic field, ROSS is the first software being developed using the open source concept, with the code being fully [...]

We do not claim that our implementation is scientifically novel, in the sense that we do not have new modeling assumptions when compared to commercial softwares.
But we believe that ROSS is a significant contribution to the available open source software and it makes research in rotordynamic faster and easier:
 - Users can write scripts for their analysis and run several simulations by automating the process. This is not possible in some commercial softwares that we have referenced, since you do not have access to the code and the only interaction with the software is through a GUI that does not have scripting tools;
- Cost is definitely a factor to make research easier, softwares such as MATLAB and ANSYS are not always available;
- We also believe that an open software with an open environment for discussion makes research easier. All the code used on the research can be shared and it is much easier to reproduce the results if everyone has access to the required tools.

Please, let me know if this answers your questions and if we need to change the paper to give a better explanation on the software contribution.

nnadeau, 2020-04-05:
Same comment as https://github.com/ross-rotordynamics/ross/issues/495#issuecomment-609410632

nnadeau, 2020-04-09:
fixed in https://github.com/ross-rotordynamics/ross/pull/517

closed 2020-04-09 by nnadeau