petrobras/ross #495: More clear Statement of Need

nnadeau, 2020-03-22:
From https://joss.readthedocs.io/en/latest/submitting.html#what-should-my-paper-contain

```
A summary describing the high-level functionality and purpose of the software for a diverse, non-specialist audience.

A clear Statement of Need that illustrates the research purpose of the software.
```

In my opinion, the paper does not fully convey *WHY* this package is needed. Is this just another simulation package?

cc https://github.com/openjournals/joss-reviews/issues/2120

raphaeltimbo, 2020-04-02:
I will add here some parts of the text that we have added/modified to make this clearer: 

_“Computational mechanical models aim to simulate the behavior of such mechanical systems. These models are used to support research and decision making. To this purpose, we present ROSS, an open source library written in Python for rotordynamic analysis.”_ 

_“To our knowledge, in the rotordynamic field, ROSS is the first software being developed using the open source concept, with the code being fully available on code hosting platforms, issues tracked online, clear license and possibility of direct contribution.”_ 

_“Developing the software using Python and its scientific ecosystem enables the user to also make use of this ecosystem, making it easier to run rotordynamics analysis. It is also easier to integrate the code into other programs, since we only use open source packages and do not depend on proprietary commercial platforms.”_

nnadeau, 2020-04-04:
@raphaeltimbo : 

- Please link the PR/commit that adds these statements
- `ROSS is the first software being developed using the open source concept`, how is that novel compared to `commercial finite element softwares that have a rotordynamic module[s]`? 
- While I personally love Python for its general ease of use, I'm not convinced by the statement `Developing the software using Python and its scientific ecosystem enables the user to also make use of this ecosystem, making it easier to run rotordynamics analysis`.

raphaeltimbo, 2020-04-04:
> @raphaeltimbo :
> 
> * Please link the PR/commit that adds these statements

Link to commit: def64b5462d227cc902cf671942101a760179270 

> * `ROSS is the first software being developed using the open source concept`, how is that novel compared to `commercial finite element softwares that have a rotordynamic module[s]`?

As discussed on #494, we believe that having an open source alternative to commercial rotordynamics softwares presents benefits to research in this area.

> * While I personally love Python for its general ease of use, I'm not convinced by the statement `Developing the software using Python and its scientific ecosystem enables the user to also make use of this ecosystem, making it easier to run rotordynamics analysis`.

My point in this statement is that with some of these rotordynamic commercial softwares you are  'locked' on their environment, interacting with the software only through a GUI, which makes it harder (impossible sometimes) to automate analysis, extract results to use in other tools, present results in different ways, run analysis with different settings and so on.
When you have an open source software developed in Python it is much easier to do those tasks since you have easy access to scientific tools, plotting tools, etc., all within the same environment.

Please, let me know if this makes the point clearer and if you want we can add it to the paper.

nnadeau, 2020-04-05:
@raphaeltimbo yes, definitely add that to the paper. `[...] 'locked' on their environment, interacting with the software only through a GUI, which makes it harder (impossible sometimes) to automate analysis, [...]` is the key point that satisfies both of my above concerns

raphaeltimbo, 2020-04-05:
@nnadeau , I have added that here: 565aee5

nnadeau, 2020-04-05:
`to automate analysis, To our knowledge, in`

typo: period instead of comma

raphaeltimbo, 2020-04-05:
> `to automate analysis, To our knowledge, in`
> 
> typo: period instead of comma

Fixed that on https://github.com/ross-rotordynamics/ross/pull/517/commits/75c93b42c24ae4eea27411349f5b517fb138b5c3

raphaeltimbo, 2020-04-08:
Hi there @nnadeau !
I forgot to tag you so I don't know if you received a notification for this.
Just wanted to let you know that I have fixed the typo.

closed 2020-04-09 by nnadeau