petrobras/ross #530: Add non synchronous force

raphaeltimbo, 2020-04-20:
We should add a function that could apply non synchronous force to the rotor.
One of the arguments of this function would be the harmonic (2x, 3x etc.).

leonardocabr, 2022-11-11:
First version of non synchronous force function implemented (Pull Request #919)

leonardocabr, 2022-11-22:
Update in function implemented (Pull Request #919)

raphaeltimbo, 2022-11-24:
Hi @leonardocabr! I will try to describe here in more details how this feature should be implemented.
In ROSS, all the analysis that we run on the model are defined by a method with the prefix 'run_'. For example, to run an unbalance response analysis we use the `.run_unbalance_response()` method like this:

```python
import ross as rs

rotor = rs.rotor_example()
speed = np.linspace(0, 1000, 101)
response = rotor.run_unbalance_response(
    node=3,
    unbalance_magnitude=10.0,
    unbalance_phase=0.0,
    frequency=speed,
)
```
The variable `response` is an object instantiated from the class `ForcedResponseResults` and it stores the results obtained after running the analyses.

For the implementation of the non synchronous force analyses I imagine something like this for the 'interface':

```python
import ross as rs

rotor = rs.rotor_example()
speed = np.linspace(0, 1000, 101)
response = rotor.run_non_synchronous_response(
    node=3,
    force_x=10.0,
    force_y=0.0,
    harmonics=[2],
    frequency=speed,
)
```
So, `force_x` would be the value of the force in the x direction in N, and harmonics would be a list from which we would then calculate the frequencies in which those forces would be applied.

To implement the method you would basically need to create an additional function/method similar to the method used by the `.run_unbalance_response()` to create an array with the forces that will be then passed to the `.run_forced_response()` method.

I would recommend taking the code from `.run_unbalance_response()` method as basis and implement it from there.

Let me know if you need more clarification and sorry for the long time to respond after you submitted the PR.

leonardocabr, 2022-11-25:
Hi @raphaeltimbo! No need to apologize! Not problem! I think that I understand the general idea. I will modify the function to a method and change the mass, damping and stiffness arrays.

closed 2024-08-22 by raphaeltimbo