petrobras/ross #532: Proportional damping

verg1lio, 2020-04-20:
This issue is intended for the discussion and decision on how (and if...) to implement the proportional damping coefficients "alpha" and "beta" on the code. 

Currently it is made as an shaft_element property, and can be chosen arbitrarily as a adjustment parameter for fine tuning the model. Another proposition is to make it a material property, and give suggestions to the user as to an "standard" range for the values.

rodrigomoliveira1, 2020-05-06:
I agree.

We could add &alpha; and &beta; parameters for both, material and shaft element. And then, adding to global damping matrix like:

![\mathbf C = \sum_{j=1}^{N_{mat}} \[\alpha_j^m \mathbf M_j + \beta_j^m \mathbf K_j\] + \sum_{i=1}^{N_{e}} \[\alpha_i^e \mathbf M_i + \beta_i^e \mathbf K_i\]](https://render.githubusercontent.com/render/math?math=%5Cmathbf%20C%20%3D%20%5Csum_%7Bj%3D1%7D%5E%7BN_%7Bmat%7D%7D%20%5B%5Calpha_j%5Em%20%5Cmathbf%20M_j%20%2B%20%5Cbeta_j%5Em%20%5Cmathbf%20K_j%5D%20%2B%20%5Csum_%7Bi%3D1%7D%5E%7BN_%7Be%7D%7D%20%5B%5Calpha_i%5Ee%20%5Cmathbf%20M_i%20%2B%20%5Cbeta_i%5Ee%20%5Cmathbf%20K_i%5D)

Where, the first sum is the proportional damping related to the material properties of each element, and the second one, related to the element itself.

RodrigoTanajura, 2022-10-25:
We are studying how (and if) we will implement this issue once again.

raphaeltimbo, 2024-03-14:
This has been implemented in the ShaftElement6DoF class.

closed 2024-03-14 by raphaeltimbo