petrobras/ross #541: Remove API name from report

raphaeltimbo, 2020-05-06:
We should remove the API name from the report module (API_Report -> report) and also make the report more general so we can apply criteria from different standards.

raphaeltimbo, 2020-06-22:
@GabrielBachmann , can you take a look at this one?

GabrielBachmann, 2020-06-30:
Sure, when we have a new version of the Report I'll remove it!

rodrigomoliveira1, 2020-06-30:
I think we should move the Report folder inside ross folder.
Then, move the api_report.py file inside Report folder. So we can keep everything related to it in the same place

GabrielBachmann, 2020-06-30:
Right now, in my remote repository there's not a API_Report folder.
The report.py file substitutes former api_report.py and there's a folder called assets in the ross.__file__ folder. 
Do you guys think that's ok, or should I create a Report folder to put assets and templates inside of it?

rodrigomoliveira1, 2020-06-30:
I think it's better to put this "assests" folder inside a "Report" folder, because as we're adding some new files and contents to this feature, it'll be more organized.

GabrielBachmann, 2020-06-30:
@rodrigomoliveira1 do you think that's good, or should I put report.py inside the report folder too? [image]

closed 2020-08-12 by GabrielBachmann