petrobras/ross #547: Bearing coefficients class

raphaeltimbo, 2020-05-11:
If we create a bearing with:
```python
stfx = 1e6
stfy = 0.8e6
bearing1 = rs.BearingElement(n=0, kxx=stfx, kyy=stfy, cxx=1e3)
```

`bearing1.kxx` will return `['1.00e+06']`, which is strange (list of strings).
Also, if the user try `bearing1.kxx[0]` it will raise:

```python
TypeError                                 Traceback (most recent call last)
<ipython-input-28-b2426d0ea068> in <module>
----> 1 bearing1.kxx[0]

TypeError: '_Stiffness_Coefficient' object is not subscriptable
```

I think we should improve the code here to a more intuitive behavior.

rodrigomoliveira1, 2020-05-13:
Each different coefficient (Kxx, Kxy, Kyx...) is an object. It returns a list of strings because of its representative method (`_Coefficient.__repr__()`). 

We have to make it an attribute instead of an object. Or define a new method that returns the coefficients as list of floats.

rodrigomoliveira1, 2020-05-13:
Actually, I found a simple way to solve this problem.

We could change this loop:
```
        for k, v in coefficients.items():
            setattr(self, k, v)
```
to this
```
        for k, v in coefficients.items():
            setattr(self, k, v.coefficient)
```

So, we have:
```
>>> b = bearing_example()
>>> b.kxx
[1000000.0,
 1000000.0,
 1000000.0,
 1000000.0,
 1000000.0,
 1000000.0,
 1000000.0,
 1000000.0,
 1000000.0,
 1000000.0,
 1000000.0]
>>> b.kxx[0]
1000000.0
```

I don't know if it's the best way yet, because we'd get an error running this command:
```
>>> b.kxx.coefficient
Traceback (most recent call last):

  File "<ipython-input-39-24bcb71d9500>", line 1, in <module>
    b.kxx.coefficient

AttributeError: 'list' object has no attribute 'coefficient'
```

commit 466f856f (2020-05-14): Change behavior for _Coefficient class

This changes the __repr__ so that we have the same repr as the
coefficients. It also adds a getitem method so that bearing.kxx[0] does
not raise an exception.

Closes #547.
closed 2020-05-14 by rodrigomoliveira1