petrobras/ross #549: BearingElement.from_table missing n_link

raphaeltimbo, 2020-05-13:
Class method `BearingElement.from_table` is missing the `n_link` argument.

rodrigomoliveira1, 2020-05-26:
We could use `**kwargs` to solve this problem and some future "sub" arguments we could implement to the class. This way, we can keep the table standards, once it only instantiate one single bearing element at time.

```
    @classmethod
    def from_table(cls, n, file, sheet_name=0, **kwargs):
        """Instantiate a bearing using inputs from an Excel table.

        A header with the names of the columns is required. These names should match the
        names expected by the routine (usually the names of the parameters, but also
        similar ones). The program will read every row bellow the header
        until they end or it reaches a NaN.

        Parameters
        ----------
        n : int
            The node in which the bearing will be located in the rotor.
        file: str
            Path to the file containing the bearing parameters.
        sheet_name: int or str, optional
            Position of the sheet in the file (starting from 0) or its name. If none is
            passed, it is assumed to be the first sheet in the file.
        kwargs : optional
            Key arguments to fill the BearingElement class. The options are:
                tag : str, optional
                    A tag to name the element
                    Default is None.
                n_link : int, optional
                    Node to which the bearing will connect. If None the bearing is
                    connected to ground.
                    Default is None.
                scale_factor : float, optional
                    The scale factor is used to scale the bearing drawing.
                    Default is 1.
                color : str, optional
                    A color to be used when the element is represented.
                    Default is '#355d7a' (Cardinal).

        Returns
        -------
        bearing: rs.BearingElement
            A bearing object.

        Examples
        --------
        >>> import os
        >>> file_path = os.path.dirname(os.path.realpath(__file__)) + '/tests/data/bearing_seal_si.xls'
        >>> BearingElement.from_table(0, file_path, n_link=1) # doctest: +ELLIPSIS
        BearingElement(n=0, n_link=1,
         kxx=array([...
        """
        parameters = read_table_file(file, "bearing", sheet_name, n)
        return cls(
            n=parameters["n"],
            kxx=parameters["kxx"],
            cxx=parameters["cxx"],
            kyy=parameters["kyy"],
            kxy=parameters["kxy"],
            kyx=parameters["kyx"],
            cyy=parameters["cyy"],
            cxy=parameters["cxy"],
            cyx=parameters["cyx"],
            frequency=parameters["frequency"],
            **kwargs,
        )
```

raphaeltimbo, 2020-05-27:
We could add the **kwargs, but I think that for these arguments (tag, n_link, scale_factor and color) we should actually list them as default keyword arguments, since they are not that many and we are already writing the docs for them.
This would also help the user when checking the function signature.

rodrigomoliveira1, 2020-05-27:
Good point, I'll do as you say.

closed 2020-05-27 by raphaeltimbo