petrobras/ross #553: Polar plot for forced response analysis

atbrandao, 2020-05-13:
The Bode plot (amplitude and phase) should provide an option to be shown in its Polar form.

In cases for which the excitation does not clearly excites a certain critical speed, the polar form of the Bode Plot can be very helpful to identify the critical speed passing.

rodrigomoliveira1, 2020-05-26:
When plotting unbalance response or frequency response, it'll be returned this format:

[image]

atbrandao, 2020-05-27:
Perfect!
The example shown generates a weird polar plot because of the low damping of the first critical. But that's exactely what I had in mind. Thanks!

closed 2020-05-27 by atbrandao