petrobras/ross #560: Ross-website link to source broken

GabrielBachmann, 2020-05-19:
When trying to open source on ross-website, it links to this [page]( https://ross-rotordynamics.github.io/ross-website/_sources/examples.rst.txt)

GabrielBachmann, 2020-05-19:
[image]

GabrielBachmann, 2020-05-19:
This is image shows the button mentioned above.
[image]

raphaeltimbo, 2020-05-22:
Hi there @GabrielBachmann !
I think this is actually the correct behavior, it shows the .rst file for that page.
To check the html generated you could use:
view-source:https://ross-rotordynamics.github.io/ross-website/examples.html

GabrielBachmann, 2020-05-22:
Oh sorry about that, I confused API and Source buttons!

closed 2020-05-22 by GabrielBachmann