petrobras/ross #569: Deflected shape diagram to unbalance response

rodrigomoliveira1, 2020-05-27:
@atbrandao, following up on your previous idea about improving the forced response (Issue #553). Would you be interested on a 3D deflected shape diagram for the rotor?
The diagram would display the calculated nodal orbits for a given `frequency`. The unbalances would be displayed as arrows on the nodes where they're applied, also indicating its phase.

The `frequency` parameter could be optional, The user could choose any frequency within the `frequency_range` argument passed to the method. The default'd be the 1st critical frequency.

@raphaeltimbo I'd like to hear your opinion too.

atbrandao, 2020-05-27:
@rodrigomoliveira1 , this would be incredibly useful! And since you came up with this idea, I'd like do go further and expand it:

- Also provide, the 2D deflected shapes showing the nodal major axis and minor axis. Some times, for complex modes, it is easier to have a physical idea of the mode shape when it is in 2D, even if it neglects the relative phases between nodes. 

- Provide a Bending Moment vs Shaft Length plot for the chosen frequency. I had to evaluate this in a case where we had severe misalignment-like forces and I was concerned about the stresses in the shaft. I had to do it on Matlab, because none of the Rotordynamic tools we use have this kind of output. A second step would be to consider surface finish and fillet radii to output the actual stresses acting on the shaft. **Maybe this should become a separate Issue....**

rodrigomoliveira1, 2020-05-27:
Good to hear that! So, I'm going to work on this feature.

> Also provide, the 2D deflected shapes showing the nodal major axis and minor axis. Some times, for complex modes, it is easier to have a physical idea of the mode shape when it is in 2D, even if it neglects the relative phases between nodes

I'll add the option for plotting 2D either.

> Provide a Bending Moment vs Shaft Length plot for the chosen frequency. I had to evaluate this in a case where we had severe misalignment-like forces and I was concerned about the stresses in the [...]

Good to know. And it's not difficult to implement this diagram. 
I'm only concerned about the code organization. For example, now we have a `.plot()` method that calls 3 other sub-methods (`plot_magnitude()`, `plot_phase()`. and `plot_polar()`) and groups the graphs like I showed in Issue #553. Adding more plots to that grid would cause the graphs to become too small and there would be a lot of information at the same time. I'll think about a solution to it, maybe I create a `plot_deflected_shape()` method to display these new graphs.

> A second step would be to consider surface finish and fillet radii to output the actual stresses acting on the shaft. Maybe this should become a separate Issue....

Yes, I guess it's something to us to think about another time.

closed 2020-06-11 by raphaeltimbo