petrobras/ross #580: Plot style

raphaeltimbo, 2020-06-08:
Since we are changing all plots to plotly, I wanted to use this opportunity to discuss how we will change the style for each plot.

One point that I have in mind is that we should not change the plots too much from the plotly default style. This will save us energy on having to write/maintain fewer lines of code and also on keeping the consistency between the styles of different plots (fonts, colors, grids etc.).

Another point is that plotly has a theme/template philosophy that is similar to that of matplotlib, so if the user needs to change the style of multiple plots for a different context (e.g. from an article to a presentation) they could do that by changing the style. As an example, when we change the theme to 'presentation' this will increase the font size for the labels to make it easier to visualize. This will not work if we change the font sizes in our plots.

Plotly has a list of themes/templates here: https://plotly.com/python/templates/ 
My suggestion is that we choose one of these themes and make the modifications to it in the top of the module, and them we should keep modifications on each plot to a minimum.

rodrigomoliveira1, 2020-06-09:
Good idea. Although I like the available templates on Plotly, we could create our own.
I've "copied and pasted" the layout parameters for all the plots, and indeed it's too much repetitive, and one can easily skip changing one of the plot styles when making modifications.

We could standarize:
- Plot Layouts:
    - title font (family, size)
    - plot background color
    - hoverlabel alignment
    - legend
        - font (family, size)
        - background color,
        - border color,
        - border width
- Axes Layout:
    - title font (family, size),
    - tick font (family, size),
     - grid color,
     - showline,
     - line width,
     - linecolor,
     - mirror,

Maybe we could also do it to `coloraxis_colorscale`, when creating ColorBars, but I don't know if it's gonna work well for Scatter colors.

raphaeltimbo, 2020-06-22:
I will start to work on a script to generate all plots for 2 or 3 different models. 
This way we will be able to evaluate how each modification affects different plots.

PR #632 (2020-07-16): Create script to generate plots
This adds a script that generates some plots for 3 different rotor
models. This helps to solve a problem we had when we changed some
property to improve the plot for a specific model, and later we would
notice that at the end the modification would be detrimental to other
models.
This, together with the plotly_theme style, closes #580.
Now we should try to make all plot modifications on the plotly_theme.py
file, and we should run the generate_comparison_plots.py script and
check the output.
After running the script, we can open the index.html file and click each [... cut]
commit 6642ff22 (2020-07-16): Add script to generate plots for comparison

This adds a script that generates some plots for 3 different rotor
models. This helps to solve a problem that we had when we modified some
property to improve the plot for a specific model and later we would
notice that at the end the modification would be detrimental to other
models.
This, together with the plotly_theme style, closes #580.
Now we should try to make all plot modifications on the plotly_theme.py
file, and we should run the generate_comparison_plots.py script and
check the output.
After running the script we can open the index.html fi [... cut]
closed 2020-07-16 by rodrigomoliveira1