petrobras/ross #589: Black - trailing comma

raphaeltimbo, 2020-06-13:
I have noticed that we have merged the code below without problems regarding the black check in our CI:
[image]

The thing that I do not undestand is that when I am running black locally it is trying to add this trailing comma. 

Is black configured in a different way in our CI? Or maybe I have some different configuration locally.

rodrigomoliveira1, 2020-06-13:
flake8 is configured to change the default max line length to 88 and to ignore the following PEP8 errors (default pattern of flake8 files on repositories):
- E203: Whitespace before ':'
- E266: Too many leading '#' for block comment 
- E501: Line too long (82 > 79 characters)
- W503: Line break occurred before a binary operator

But, there's no instructions related to checking for trailing commas (no warnings are raised).

About Black, it runs with default options. I think we didn't change any option from it.
Look how Black deals with trailing commas [here](https://black.readthedocs.io/en/stable/the_black_code_style.html#trailing-commas). It says it will remove trailing commas if the arguments fit in one single line, which is the case in this example. So, I think that's why travis won't complain about the missing commas at the end of the line. Maybe it's some different local configuration.

Actually, I have already noticed that, sometimes, some commits adds a trailing comma and others removes them.

raphaeltimbo, 2020-06-13:
> About Black, it runs with default options. I think we didn't change any option from it.
> Look how Black deals with trailing commas [...]

But since Black removes the trailing commas, travis should have failed in this example.
This is the PR related to the example:
https://github.com/ross-rotordynamics/ross/pull/574/files
And here we have the travis build:
https://travis-ci.com/github/ross-rotordynamics/ross/jobs/347282600

Black passes without problem:
[image]

rodrigomoliveira1, 2020-06-13:
You are right. 
And it's odd, because it runs fine with and without commas.

rodrigomoliveira1, 2020-06-19:
I think this is also an issue raised in Black repo. They are discussing pretty much the same thing. Look this [issue](https://github.com/psf/black/issues/1288).

raphaeltimbo, 2020-06-19:
I think you found the problem @rodrigomoliveira1 . It looks like they have introduced this in the last release:
[image]
And I am using the 19.3 release, so maybe that could be the problem.
I will update my version of black now and I think we can close this issue for now.

closed 2020-06-19 by raphaeltimbo