petrobras/ross #590: Doctests creates several files

raphaeltimbo, 2020-06-14:
When we run the doctests they create several files in the ROSS directory, this is annoying since these documents can sometimes go into commits by mistake.
Ideally doctests would not create or modify files.
I was looking for solutions to this in numpy docstrings and I saw that they use a temporary file in this case:
```python
    Examples
    --------
    >>> from tempfile import TemporaryFile
    >>> outfile = TemporaryFile()
    >>> x = np.arange(10)
    >>> np.save(outfile, x)
```

Another option would be to add these files to a /tmp/ folder that could be added to the project's .gitignore file:
```python
    Examples
    --------
    Store data to disk, and load it again:
    >>> np.save('/tmp/123', np.array([[1, 2, 3], [4, 5, 6]]))
    >>> np.load('/tmp/123.npy')
    array([[1, 2, 3],
           [4, 5, 6]])
```

rodrigomoliveira1, 2020-06-14:
The second solution seens better to me too.

raphaeltimbo, 2020-07-02:
I have implemented the second solution and we have merged the code in #601. 
The problem is that appveyor was not working at that time (#611) and we didn't notice that the code does not work on windows.
I will implement the solution based on the tempfile module (https://stackoverflow.com/a/847866/5726899).

PR #601 (2020-06-27): Change save functions
This PR makes some modifications to the save functions implemented in each element class:

- [x] Rotor will be saved to a single `.toml` file. 
Each element has a .save method that writes to a .toml file, if the
file already exists the method appends to the end of the file. So the
process of saving the rotor is basically an iteration over all the
rotor elements and a call to the element.save() method.
The load method goes through all items in the .toml file and calls the
elements .read_toml_data, which parses the dictionary stored in the
.toml and returns the Element.
The [... cut]
PR #612 (2020-07-02): Change tempfile to be crossplatform
The approach used before ('/tmp/file') would work only on unix systems
since this path does not exist on windows.
Now the tempdir is obtained using the tempfile module.

Closes #590.
closed 2020-07-02 by rodrigomoliveira1