petrobras/ross #592: Add probes to the rotor model

raphaeltimbo, 2020-06-15:
We should add an object called `Probe` to ROSS.
The user would instantiate this object with arguments `(node, probe_orientation)` and pass it to the rotor constructor. 
The object would be used in the `plot_rotor()` method and we should also be able to use it as an argument in methods such as plot frequency response.

rodrigomoliveira1, 2020-07-06:
We could, also, modify some methods that requires a `dof` input to a `(node, probe_orientation)` input like you said.

rodrigomoliveira1, 2020-07-14:
@raphaeltimbo, I'm adding this feature to the rotordynamic report. 
Is it important to add probes as object to the `Rotor`, like the elements? I was leaning to add this option to `.plot()` methods. I see this way more flexible to the user.

For example:

- probes argument: inputting multiple tuples `(probe_node, probe_orientation)`
```python
>>> import ross as rs
>>> rotor = rs.rotor_example()
>>> speed = np.linspace(0, 1000, 101)
>>> response = rotor.run_freq_response(speed_range=speed)
>>> response.plot(probes=[(node1, orientation1), (node2, orientation2), ... ])
```
OR
- `probe_node` and `probe_orientation` arguments
```python
>>> import ross as rs
>>> rotor = rs.rotor_example()
>>> speed = np.linspace(0, 1000, 101)
>>> response = rotor.run_freq_response(speed_range=speed)
>>> response.plot(probe_node =[node1, node2, ...], probe_orientation =[orientation1, orientation2, ...])
```

In both cases we could use pint.Quantity.

What's your opinion?

raphaeltimbo, 2020-07-15:
I agree with you that it is more flexible this way.
I prefer the first option: `probes=[(node1, orientation1), ...]`

rodrigomoliveira1, 2020-07-21:
I had to do some changes on the code to make it work.

- `_calculate_major_axis()` method
First, to calculate the displacements for different orientions than 0º (x dir.) and 90º (y dir.), I built a slightly different method to calculate the semi axes. This method calculates the semi axes for all speed values in the `speed_range` argument, given a single node.
The other one, which is already used to calculate the deflected shape, calculates the semi axes for all the nodes, given single speed value.
    - Alternative:
Change the current `_calculate_major_axis()` method to receive a single speed value and a single node value. The plot methods would iterate over it (through speed values or node values) and fill the arrays that would be created in the plot methos instead of being created in `_calculate_major_axis()`.
In this case, `.plot()` method iterates over speed values (with node fixed) and `plot_deflected_shape()` iterates over nodes values (with speed fixed).

- `plot()` method
`.plot()` method now gets a `probe` argument. It must be a list of tuples, where each tuple contains the node and orientation for the probe.
`.plot(probe = [(node1, orient1), (node2, orient2), ...])`
This change, allows the user to plot several results on the same chart, as you can see below:

[image]

rodrigomoliveira1, 2020-07-23:
It was not necessary to add a new method. The displacement measured by the `probe` argument is calculated by and operator applied to the `probe_resp (X, Y)` coordinates, which rotates the vector according to the probe orientation `angle`. Then, the absolute value is obtained by the equation in polar coordinates for the elipse.

```python
        for i, p in enumerate(probe):
            dofx = p[0] * number_dof
            dofy = p[0] * number_dof + 1
            angle = Q_(p[1], probe_units).to("rad").m

            operator = np.array(
                [[np.cos(angle), - np.sin(angle)],
                 [np.cos(angle), + np.sin(angle)]]
            )

            probe_resp = operator @ np.vstack((magnitude[dofx], magnitude[dofy]))
            probe_disp = np.sqrt((probe_resp[0] * np.cos(angle)) ** 2 +
                                 (probe_resp[1] * np.sin(angle)) ** 2)
```

closed 2020-07-24 by raphaeltimbo