petrobras/ross #611: AppVeyor not working

rodrigomoliveira1, 2020-07-02:
AppVeyor CI has not been working properly lately.
This message is appearing in every commit we have done recently.

[image]

The [build](https://ci.appveyor.com/project/raphaeltimbo/ross/builds/33602334) which began to fail.

rodrigomoliveira1, 2020-07-02:
I found a suggestion saying GitHub authorization is the most probable reason for this. since we did not change anything in appveyor.yml recently. Maybe appveyor.yml cannot be read. So, re-authorizing GitHub at https://ci.appveyor.com/account, should help.

raphaeltimbo, 2020-07-02:
I have tried this now and restarted the last build. Lets see if it works.

rodrigomoliveira1, 2020-07-02:
It works. 
But now we have to raise another Issue about appveyor and the temp files used to test the save methods. I think appveyor is not dealing well with them

raphaeltimbo, 2020-07-02:
I have reopened #590 to work on this.

closed 2020-07-02 by raphaeltimbo