petrobras/ross #622: Campbell - Critical speeds markers not working properly

raphaeltimbo, 2020-07-08:
The critical speed markers are not working properly for the Campbell diagram.

[image]

Finding the exact spot for the intersection of the curves is something that we also need for the UCS, where we need to find the intersection between the mode and the bearing stiffness. Maybe we should create a utility function to use in both cases. 
The approach used in the UCS now looks costly to compute.
I am thinking about something similar to what we have in the Campbell diagram now (seeing change in signs to check where the lines have crossed) mixed with linear interpolation:
1. First find 4 closest points (one pair for each curve) that represent the intersection for the curves;
2. Assemble a linear system to calculate the exact intersection coordinates (x, y).

rodrigomoliveira1, 2020-07-09:
It searches only once per natural frequency column (from the frequency array). That's why, even the harmonic crossing twice the same natural frequency curve, it displays only the first mark.

The other bug, related to incorrect mark spot, must be related to the results beahviour. I think it was trying to inteporlate between `wd=0` and `wd=25k`, So, the virtual points were closer to the harmonics in those `x` spots.

raphaeltimbo, 2020-07-13:
I found an interesting code here: https://github.com/sukhbinder/intersection
Think I will add it in our utils module. This way we will be able to use it with the Campbell and UCS.

rodrigomoliveira1, 2020-07-13:
I did a little change to the code, and now it marks multiple interceptions for the same natural frequency curve.
The problem I see here, is the behaviour of this map in particular.

[image]

The scatters which are mostly red trick us. These 2 curves seems to be "continuos" lines that starts at 800rad/s (for speed = 0) and ends at 700rad/s (for speed = 3000). But, actually, they are different natural frequencies. Maybe the figure below helps to understand what's happening.

[image]

Again, due the interpolation we have added, it makes sense the it's marking critical speeds at 200rad/s (~2000RPM).

Anyway, I'd like to test the code you've suggested.

raphaeltimbo, 2020-07-13:
@rodrigomoliveira1 , as you mentioned, the marking at 200 rad/s is something specific to this map.
The modification from PR #628 gives the same result as you have posted here.

rodrigomoliveira1, 2020-07-13:
This is the modified code I mentioned.
```python
        critx = np.array([])
        crity = np.array([])

        scatter_marker = ["triangle-up", "circle", "triangle-down"]
        for mark, whirl_dir, legend in zip(
            scatter_marker, [0.0, 0.5, 1.0], ["Foward", "Mixed", "Backward"]
        ):
            for i in range(num_frequencies):
                w_i = wd[:, i]
                whirl_i = whirl[:, i]
                log_dec_i = log_dec[:, i]

                for harm in harmonics:
                    idx = np.argwhere(np.diff(np.sign(w_i - harm * speed_range)))
                    idx = idx.flatten()

                    if len(idx) != 0:
                        for ix in idx:
                            interpolated = interpolate.interp1d(
                                x=[speed_range[ix], speed_range[ix + 1]],
                                y=[w_i[ix], w_i[ix + 1]],
                                kind="linear",
                            )
                            xnew = np.linspace(
                                speed_range[ix],
                                speed_range[ix + 1],
                                num=30,
                                endpoint=True,
                            )
                            ynew = interpolated(xnew)
                            ix_ = np.argwhere(
                                np.diff(np.sign(ynew - harm * xnew))
                            ).flatten()

                            critx = np.append(critx, xnew[ix_])
                            crity = np.append(crity, ynew[ix_])

                whirl_mask = whirl_i == whirl_dir
                if whirl_mask.shape[0] == 0:
                    continue
                else:
                    fig.add_trace(
                        go.Scatter(
                            x=speed_range[whirl_mask],
                            y=w_i[whirl_mask],
                            marker=dict(
                                symbol=mark,
                                color=log_dec_i[whirl_mask],
                                coloraxis="coloraxis",
                            ),
                            mode="markers",
                            name=legend,
                            legendgroup=legend,
                            showlegend=False,
                            hoverinfo="none",
                        )
                    )

        fig.add_trace(
            go.Scatter(
                x=critx,
                y=crity,
                mode="markers",
                marker=dict(symbol="x", color="black"),
                name="Crit. Speed",
                legendgroup="Crit. Speed",
                showlegend=False,
                hovertemplate=(
                    "Frequency: %{x:.2f}<br>"
                    + "Critical Speed: %{y:.2f}"
                ),
            )
        )
```

commit 7166954e (2020-07-14): Add intersection function to utils module

The intersection function can be used to find the intersection point
between two curves. The Campbell diagram and the ucs plot had some
implementation for that, so it is useful to add this single function
to the utils module so that these functions can use the same approach
to find the intersection point.

Closes #622.
closed 2020-07-14 by rodrigomoliveira1