petrobras/ross #637: Tutorial - texts with docstrings

raphaeltimbo, 2020-07-27:
In our tutorial we have several cells in markdown that have the docstring for a function in it. Example:
[image]

One problem that this can cause is that we have to update these cells by hand if we change some docstring, eventually we will miss some updates.

The other problem that I see here is that our tutorial seems very long because we are putting these texts in lots of places.

Another point is that, since the tutorial is in a jupyter notebook, it is very easy to the user to get the documentation with `?` or <shift + tab> inside the parenthesis, so I don't think we gain much by writing everything again in a cell.

So, I would like to propose that we remove these docstrings in the tutorial.

rodrigomoliveira1, 2020-08-04:
I'm working on these changes. By the way, without the docstrings, users will need to access the API page on ROSS' website, so they get a complete information.
With that in mind, I'd suggest we add other files to the API page, like point_mass.py, results.py, fluidflow module and stochastic module.

rodrigomoliveira1, 2020-08-26:
PR #653 closes this issue.

closed 2020-08-26 by rodrigomoliveira1