petrobras/ross #674: Tutorial links not working

raphaeltimbo, 2020-09-22:
I tried to use the links for the tutorial inside the ross web page (https://ross-rotordynamics.github.io/ross-website/examples/tutorial.html) and, although they work when we are using the jupyter notebook, it looks like they do not work there on the website.

rodrigomoliveira1, 2020-09-23:
The funny part is it's a HTML command. It was supposed to work on a website too.

gutilacerda, 2022-10-10:
In the ross current website the pages of tutorial and examples are working. Is there another problem related with the website?

raphaeltimbo, 2022-10-11:
Hi @gutilacerda.
The problem here in this issue is that, in the image bellow, the links in the blue part do work but the links at the red part do not work. 
Here we have some options:
 - Remove the 'Table of Contents' in the red part and stay only with the 'Contents' in the blue part. This has the down side of not having the links when the users are running the notebook on their machines;
 - Figure out a way to make these links work;
 - Try to remove this 'Table of Contents' from the online documentation only, maybe by adding some tag to the notebook cell metadata (we would need to check if [myst-nb](https://myst-nb.readthedocs.io/en/latest/) has this feature).

raphaeltimbo, 2022-10-11:
Sorry, @gutilacerda. There was some error when I pasted in the message.
I will edit the message and add it.

closed 2023-03-14 by raphaeltimbo