petrobras/ross #675: Deflected shape plot layout

raphaeltimbo, 2020-09-22:
The deflected shape 3d plot does not have enough space in the current layout:
[image]

Maybe we could move the legend to below the plot, like we have in the Campbell diagram.

rodrigomoliveira1, 2020-09-23:
Like this one:

[image]

rodrigomoliveira1, 2020-09-23:
The overlapping labels remais an issue. A simple solution is setting a default size for this figure (800px x 600px - 4/3 ratio).
The figures gets slightly larger but has a much better layout.

[image]

I tried doing 640px x 480px, but it didn't fit well.

[image]

rodrigomoliveira1, 2020-09-23:
I did some domain and camera work to set a better viewpoint to the 3D plot

[image]

rodrigomoliveira1, 2020-09-28:
PR #676 closes this issue.

closed 2020-09-28 by rodrigomoliveira1