petrobras/ross #677: Examples not working

Antannah, 2020-09-24:
I tried to run the examples in /doce/examples/, and got 

    TypeError: __init__() got an unexpected keyword argument 'sparse'

when rs.Rotor is called in function create_rotor(n_el).
```
def create_rotor(n_el):
  ...
  return rs.Rotor(shaft, disks, bearings, sparse=False)
```
Looks like rs.Rotor doesn't know the keyword sparse. Sparse is also not mentioned in the APIreference as well as it's not in the source.
taking sparse out and changing to `return rs.Rotor(shaft, disks, bearings)` gives
```

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-7-ef8aacbe2e82> in <module>
     10     rotor = create_rotor(n_el, R)
     11     modal = rotor.run_modal(speed=0)
---> 12     errors[i, :] = abs(
     13         100 * (modal.wn[:n_eigen] - modal_80.wn[:n_eigen])
     14         / modal_80.wn[:n_eigen])

ValueError: could not broadcast input array from shape (6) into shape (8)
```
What can I do? Thank's in advance.
Antannah

rodrigomoliveira1, 2020-09-25:
Hello, @Antannah thanks for your comments. We have released v0.4.0 recently, but we didn't updated the examples. Sorry about that. We're going to fix it very soon.

The `sparse` argument has been move to `run_modal()` method, since it makes more sense to pick this option when running a modal analysis than picking it when building the rotor model (see Issue #573 and #583).

Here is this example with some changes to work properly:

```python
import numpy as np
import matplotlib.pyplot as plt
import ross as rs

steel = rs.materials.steel

number_of_elements = [2, 3, 4, 5, 6, 7, 8]

def create_rotor(n_el, R, shear_effects=False, rotary_inertia=False):
    """Create example rotor with given number of elements and R ration."""
    L_total = 1
    D = R * L_total
    
    shaft = [
        rs.ShaftElement(1./(n_el), idl=0, odl=D, material=steel,
                        shear_effects=shear_effects,
                        rotary_inertia=rotary_inertia)
        for i in range(n_el)
    ]

    bearings = [
        rs.BearingElement(0, kxx=1e15, cxx=0),
        rs.BearingElement(n_el, kxx=1e15, cxx=0)
    ]
    
    return rs.Rotor(shaft_elements=shaft, bearing_elements=bearings)

# create reference rotor with 80 elements
R = 0.04
rotor_80 = create_rotor(80, R)
modal_80 = rotor_80.run_modal(speed=0)

n_eigen = 6
errors = np.zeros([len(number_of_elements), n_eigen])

for i, n_el in enumerate(number_of_elements):
    rotor = create_rotor(n_el, R)
    modal = rotor.run_modal(speed=0, num_modes=2*n_eigen)
    errors[i, :] = abs(
        100 * (modal.wn[:n_eigen] - modal_80.wn[:n_eigen])
        / modal_80.wn[:n_eigen])

fig, ax = plt.subplots()
ax.set_xlabel('number of elements')
ax.set_ylabel('Natural Frequency error(%)')
for i in range(n_eigen):
    ax.semilogy(number_of_elements, errors[:, i])

# create reference rotor with 80 elements
n_el = 6 
R_list = np.linspace(1e-4, 0.15, 10)

n_eigen = 6
errors = np.zeros([len(R_list), n_eigen])

for i, R in enumerate(R_list):
    rotor_ref = create_rotor(100, R, 
                             shear_effects=True, rotary_inertia=True)
    modal_ref = rotor_ref.run_modal(speed=0)

    rotor = create_rotor(n_el, R)
    modal = rotor.run_modal(speed=0)
    errors[i, :] = abs(
        100 * (modal.wn[:n_eigen] - modal_ref.wn[:n_eigen]) 
        / modal_ref.wn[:n_eigen])

fig, ax = plt.subplots()
ax.set_xlabel('Slenderness ratio')
ax.set_ylabel('Natural Frequency error(%)')
for i in range(n_eigen):
    ax.semilogy(R_list, errors[:, i])
```

rodrigomoliveira1, 2020-09-25:
@Antannah
We have updated our examples and done a bump version to 0.4.1, fixing some bugs. Try updating your ROSS version or downloading it directly from Github, and let us know if something else is not working.

Antannah, 2020-09-26:
@rodrigomoliveira1 @GabrielBachmann

Works fine now, Thanks alot!!!
Issue seems to bee closed in my eyes.
Best regards 
Antannah

closed 2020-09-26 by Antannah