petrobras/ross #686: Unexpected errors in `generate_comparasion_plot.py`

rodrigomoliveira1, 2020-10-08:
In PR #652, @verg1lio, @bela-bs and @marcusfilipesr are experiencing some unexpected errors with `generate_comparasion_plot.py` while testing the new defect module.

We could add some routines to check the correct functioning of this file.

@verg1lio, it would be very useful if you could post here what exactly is happening and the log error.

verg1lio, 2020-10-08:
Hello @rodrigomoliveira1 , good afternoon! The explanation should be uploaded in a moment.

marcusfilipesr, 2020-10-08:
Hello everyone, so here is the log error that you are asking for. This log error occurs when the pytest library tries to run the `generate_comparison_plot.py`.
```
============================= test session starts ==============================
platform linux -- Python 3.6.9, pytest-6.1.0, py-1.9.0, pluggy-0.13.1
rootdir: /home/izabela/Documents/Projeto EDGE Petro/ross, configfile: pytest.ini
collected 0 items / 2 errors
==DISCOVERED TESTS BEGIN==
{"tests": [], "errors": [{"file": "tools/generate_comparison_plots.py", "message": "tools/generate_comparison_plots.py:85: in <module>\n    for rotor in rotors\ntools/generate_comparison_plots.py:85: in <dictcomp>\n    for rotor in rotors\nross/results.py:1624: in plot\n    probe, probe_units, frequency_units, amplitude_units, **mag_kwargs\nross/results.py:1321: in plot_magnitude\n    for i, p in enumerate(probe):\nE   TypeError: 'int' object is not iterable"}, {"file": "tools/generate_comparison_plots.py", "message": "tools/generate_comparison_plots.py:85: in <module>\n    for rotor in rotors\ntools/generate_comparison_plots.py:85: in <dictcomp>\n    for rotor in rotors\nross/results.py:1624: in plot\n    probe, probe_units, frequency_units, amplitude_units, **mag_kwargs\nross/results.py:1321: in plot_magnitude\n    for i, p in enumerate(probe):\nE   TypeError: 'int' object is not iterable"}]}
==DISCOVERED TESTS END==

 

==================================== ERRORS ====================================
_____________ ERROR collecting tools/generate_comparison_plots.py ______________
tools/generate_comparison_plots.py:85: in <module>
    for rotor in rotors
tools/generate_comparison_plots.py:85: in <dictcomp>
    for rotor in rotors
ross/results.py:1624: in plot
    probe, probe_units, frequency_units, amplitude_units, **mag_kwargs
ross/results.py:1321: in plot_magnitude
    for i, p in enumerate(probe):
E   TypeError: 'int' object is not iterable
_____________ ERROR collecting tools/generate_comparison_plots.py ______________
tools/generate_comparison_plots.py:85: in <module>
    for rotor in rotors
tools/generate_comparison_plots.py:85: in <dictcomp>
    for rotor in rotors
ross/results.py:1624: in plot
    probe, probe_units, frequency_units, amplitude_units, **mag_kwargs
ross/results.py:1321: in plot_magnitude
    for i, p in enumerate(probe):
E   TypeError: 'int' object is not iterable
=============================== warnings summary ===============================
ross/rotor_assembly.py:1041
ross/rotor_assembly.py:1041
  /home/izabela/Documents/Projeto EDGE Petro/ross/ross/rotor_assembly.py:1041: UserWarning:
  
  Extrapolating bearing coefficients. Be careful when post-processing the results.

 

-- Docs: https://docs.pytest.org/en/stable/warnings.html
=========================== short test summary info ============================
ERROR tools/generate_comparison_plots.py - TypeError: 'int' object is not ite...
ERROR tools/generate_comparison_plots.py - TypeError: 'int' object is not ite...
!!!!!!!!!!!!!!!!!!! Interrupted: 2 errors during collection !!!!!!!!!!!!!!!!!!!!
================== 2 warnings, 2 errors in 231.02s (0:03:51) ===================
```
As we (@verg1lio , @bela-bs , @hiagopinacio , and @marcusfilipesr ) were looking to solve the test problems, we found two problems. The first one is that in this file `\tools\` it can not load the `ross` package as it follows on the .py file's preamble.
```python
from pathlib import Path

import numpy as np
import plotly.io as pio
from jinja2 import Environment, FileSystemLoader

import ross as rs # This import
```

To fix this problem, we added to the preamble the following lines:
```python
import sys
import os.path

sys.path.append(
    os.path.abspath(os.path.join(os.path.dirname(__file__), os.path.pardir))
)
```

This fixed the first problem.

=================

The second problem was a little bit more difficult to find. This one was located at lines 72 - 86, as it follows:
```python
kwargs = {
    rotor: pio.to_html(
        rotors[rotor]
        .run_unbalance_response(
            node=0,
            unbalance_magnitude=0.5,
            unbalance_phase=0,
            frequency=np.linspace(0, 2000),
        )
        .plot(0),
        include_plotlyjs=False,
        full_html=False,
    )
    for rotor in rotors
}
```
When the pytest executed those lines it calls the `roto.run_unbalance_response()` method, which calls the method `rotor.forced_response()` which calls the class `ForcedResponseResults()` in the `results.py` file. On the lines above, the method `.plot(0)` is also used. But when we were checking this method on the `ForcedResponseResults()` class, we found that the first required parameter for this method was the probe properties, which has to be a list. This `.plot(0)` was sending `0` as a probe and it was breaking the code on the 1321 line, which is:
```python
 for i, p in enumerate(probe): # This for was trying to read a int parameter
            dofx = p[0] * number_dof
            dofy = p[0] * number_dof + 1
            angle = Q_(p[1], probe_units).to("rad").m
...
```
To fix this, we changed the method `.plot(0)` to `.plot([(0,0)])`, and now the pytest does not show any errors on this file.

==============

**We did these changes locally, so we did not commited those yet.**

rodrigomoliveira1, 2020-10-08:
Thank you very much, @marcusfilipesr. That was enlightening.
The `plot(0)` error makes sense for what you have explained, and what you are proposing to solve is correct.
We didn't update this file with the changes done to v0.4.0, that's the reason for these errors.

We'd be thankful if you commit these changes too.

Adding tests to `generate_comparison_plot.py` will prevent these problems from happening again.

marcusfilipesr, 2020-10-08:
You're welcome @rodrigomoliveira1, we are pleasured to know that we've contributed to fix this bug.
About the commit, would you like us to commit on the PR #652 or in another just to fix this particular issue?

rodrigomoliveira1, 2020-10-08:
I think a new PR would fit better.

verg1lio, 2020-10-08:
@rodrigomoliveira1 , @bela-bs just committed the changes now at 

https://github.com/ross-rotordynamics/ross/pull/652/commits/03888d75f335fc01fb1af01bd58e98aa0f485a45

marcusfilipesr, 2020-10-08:
I think the changes commited were just about the misalignment class. So we still need the new PR to make the changes discussed here, or am I wrong?

rodrigomoliveira1, 2020-10-08:
if those commit did not fix this issue, yes.

commit 268cfcd6 (2020-10-14): Merge pull request #689 from LMEst-Rotor/fix_generate

Fix #686
closed 2020-10-14 by raphaeltimbo