petrobras/ross #693: TypeError: can't multiply sequence by non-int of type 'numpy.float64'

nilswagner, 2020-10-21:
File "example2.py", line 120, in <module>
    show(rotor_list[0].plot_rotor(plot_type='matplotlib'))
  File "/home
[example2.py.gz](https://github.com/ross-rotordynamics/ross/files/5416940/example2.py.gz)
/wagner/git/ross/ross/rotor_assembly.py", line 1814, in plot_rotor
    step = disk.scale_factor * mean_od
TypeError: can't multiply sequence by non-int of type 'numpy.float64'

rodrigomoliveira1, 2020-10-21:
Hello @nilswagner! Thank you for reporting this issue. 
Could you inform us the ROSS version you are using? We have released version 0.4.1 a couple of weeks ago, fixing some problems regarding to examples (that were out to date).

By the command you are informing `show(rotor_list[0].plot_rotor(plot_type='matplotlib'))` I assume you are using version 0.3 or 0.4.0. 

If it's the case, try updating the package by using `pip install ross-rotordynamics -U`

Tell us if there's anything we could help you out.

raphaeltimbo, 2020-12-09:
I am assuming everything is ok here, so I am going to close the issue.
Feel free to reopen it.

closed 2020-12-09 by raphaeltimbo