petrobras/ross #699: Can't make bearing supports work

LeifParr, 2020-11-13:
Hello all.

I've been doing rotordynamic calculations of mainly hydropower units for more than 10 years, and I recently came across ROSS for Python. We use the RAPPID package for our rotordynamic calculations, but the plotting is not very nice, and it lacks some flexibility for custom analyses.

I've started using it for simple cases such as plotting critical speeds across a range of bearing stiffnesses. Now I'm trying to move on to a more detailed application, where I want to add point mass nodes and use n_link to be able to model bearing housing mass and stiffness.

I tried running the example from the tutorial, which works for plotting the rotor, but fails when I try to run any analysis. I think I have tracked it down to the fact that the GlobalIndexes of the point mass elements are not integers. I'm kind of stuck as to how to get around this.

This is the code, pretty much copied from the tutorial, except I am trying to run a modal analysis.
`# Example from tutorial
import ross as rs
import numpy as np
n = 6
steel = rs.materials.steel

shaft_elem = [
    rs.ShaftElement(
        L=0.25,
        idl=0.0,
        odl=0.05,
        material=steel,
        shear_effects=True,
        rotary_inertia=True,
        gyroscopic=True,
    )
    for _ in range(n)
]

disk0 = rs.DiskElement.from_geometry(
    n=2, material=steel, width=0.07, i_d=0.05, o_d=0.28
)
disk1 = rs.DiskElement.from_geometry(
    n=4, material=steel, width=0.07, i_d=0.05, o_d=0.28
)
disks = [disk0, disk1]

stfx = 1e6
stfy = 0.8e6
my_node = int(7)
bearing0 = rs.BearingElement(0, kxx=stfx, kyy=stfy, cxx=0, n_link=my_node)
bearing1 = rs.BearingElement(6, kxx=stfx, kyy=stfy, cxx=0)
bearing2 = rs.BearingElement(my_node, kxx=stfx, kyy=stfy, cxx=0)

bearings = [bearing0, bearing1, bearing2]

pointmass = [rs.PointMass(n=my_node, m=30)]

rotor1 = rs.Rotor(shaft_elem, point_mass_elements=pointmass, disk_elements=disks, bearing_elements=bearings)
for my_elem in rotor1.elements:
    print(my_elem.dof_global_index)

rotor1.plot_rotor().show()
my_modal=rotor1.run_modal(10)`

Here's the output and error message. Notice that the global-indexes are non-integer for the bearing element with n_link, and the point mass.

GlobalIndex(x_0=0, y_0=1, alpha_0=2, beta_0=3, x_1=4, y_1=5, alpha_1=6, beta_1=7)
GlobalIndex(x_1=4, y_1=5, alpha_1=6, beta_1=7, x_2=8, y_2=9, alpha_2=10, beta_2=11)
GlobalIndex(x_2=8, y_2=9, alpha_2=10, beta_2=11, x_3=12, y_3=13, alpha_3=14, beta_3=15)
GlobalIndex(x_3=12, y_3=13, alpha_3=14, beta_3=15, x_4=16, y_4=17, alpha_4=18, beta_4=19)
GlobalIndex(x_4=16, y_4=17, alpha_4=18, beta_4=19, x_5=20, y_5=21, alpha_5=22, beta_5=23)
GlobalIndex(x_5=20, y_5=21, alpha_5=22, beta_5=23, x_6=24, y_6=25, alpha_6=26, beta_6=27)
GlobalIndex(x_2=8, y_2=9, alpha_2=10, beta_2=11)
GlobalIndex(x_4=16, y_4=17, alpha_4=18, beta_4=19)
GlobalIndex(x_0=0, y_0=1, x_7=28, y_7=29)
GlobalIndex(x_6=24, y_6=25)
GlobalIndex(x_7=28.0, y_7=29.0)
GlobalIndex(x_7=28.0, y_7=29.0)
Traceback (most recent call last):
  File "C:/Leif/Python_L/Rotordynamics/Example_Tutorial_RotorBearingHousing.py", line 44, in <module>
    my_modal=rotor1.run_modal(10)
  File "C:\Program Files\Python38\lib\site-packages\ross\rotor_assembly.py", line 605, in run_modal
    evalues, evectors = self._eigen(speed, num_modes=num_modes, sparse=sparse)
  File "C:\Program Files\Python38\lib\site-packages\ross\rotor_assembly.py", line 1164, in _eigen
    A = self.A(speed=speed, frequency=frequency)
  File "C:\Program Files\Python38\lib\site-packages\ross\rotor_assembly.py", line 973, in A
    np.hstack([la.solve(-self.M(), self.K(frequency)), la.solve(-self.M(), (self.C(frequency) + self.G() * speed))])])
  File "C:\Program Files\Python38\lib\site-packages\ross\rotor_assembly.py", line 841, in M
    M0[np.ix_(dofs, dofs)] += elm.M()
IndexError: arrays used as indices must be of integer (or boolean) type

All the best,
Leif Parr
Mechanical engineer

Rotor plot:
[image]

rodrigomoliveira1, 2020-11-13:
Hello @LeifParr, thank you for reporting this bug to us.
Through you error log, the `GlobalIndex` values getting non integer values is causing this problem, like you've saud.
I'm going to fix it soon.

rodrigomoliveira1, 2020-11-13:
@LeifParr,

PR #700 should fix this issue.
While this update is not released in a new version, please, download the file directly from GitHub and replace it inside `ross` folder. If you have `Anaconda` installed, search for the path `... Anaconda3\Lib\site-packages\ross`.

Let us know if you have any other issue with ROSS. We are glad for you to try our software. Suggestions for improvements are also welcome.

LeifParr, 2020-11-14:
Good evening. I am quite impressed by fast feedback on my question above. I tried with your fix, and am now able to run with support for all three bearings, see plot below:

[image]

This means that I can continue exploring and using the ROSS library in parallell with the RAPPID package. Thank you.

closed 2020-11-14 by LeifParr