petrobras/ross #710: Unbalance response magnitude units is 0-peak?

LeifParr, 2020-12-01:
I continue to explore the excellent ROSS library. I have run through a fairly complex hydropower unit, with three bearings with housings, and compared damped natural frequencies with the rotordynamic software RAPPID with good agreement between results.
I now moved on, to compare forced response results. It seems to me that the output I get from the ross rotor.run_unbalance_response() function is in unit 0-peak. 
The rotor is like so:
[image]
The Ross response in DE bearing (probe 2) is about 0.2 mm, while RAPPID response is about 0.4 mm peak-peak. The same goes for the other nodes, I therefore assume that the forced output is in 0-peak.
Is it possible to add a note about this in the tutorial and/or at appropriate location in the code itself? I tried looking through the code for clues.

[image]

[image]

raphaeltimbo, 2020-12-01:
Hi there @LeifParr ! 
Yes, the output in this case is 0-peak, and you are right that our documentation is not clear about this.
I will update the docs to reflect this.
Also, in this plot we have the option to change the units with `amplitude_units=<units>`. 
I will try to add to our unit registry the option for pk-pk, since this is such a common use in rotordynamics.

leoishimoto, 2020-12-02:
I never tested the forced response, but, in any case, I want to add some comments about display of the results.
The major and minor axis of the ellipse is highly desirable not only the probes reference. In circular orbits does not make difference but for anisotropic bearings is necessary.
For rolling element bearing’s the response in velocity / acceleration is also desired including the RMS values making the comparison more appropriate with field results.

rodrigomoliveira1, 2020-12-02:
>I never tested the forced response, but, in any case, I want to add some comments about display of the results.
The major and minor axis of the ellipse is highly desirable not only the probes reference. In circular orbits does not make difference but for anisotropic bearings is necessary.
For rolling element bearing’s the response in velocity / acceleration is also desired including the RMS values making the comparison more appropriate with field results.

I'm solving this topic right now. As our current API requires `probes=(node, orientation)` to locate where the user wants to "read" the displacement, I'm inserting an option to the orientation that allows the user to get the major and minor axes too.
So, for an example:
```python
>>> import ross as rs

>>> # RUN UNBALANCE RESPONSE
>>> rotor = rs.rotor_example()
>>> speed = np.linspace(0, 1000, 201)
>>> response = rotor.run_unbalance_response(
...                                         node=3,
...                                         unbalance_magnitude=0.001,
...                                         unbalance_phase=0.0,
...                                         frequency=speed)

>>> # POST-PROCESSING
>>> # Get results from node 3 with 45° orientation
>>> probe_node_1 = 3
>>> probe_angle_1 = np.pi / 4

>>> # Get results from node 3 for major axis
>>> probe_node_2 = 3
>>> probe_angle_2 = "max" # here we insert the string "max" to get the major axis values

>>> # Get results from node 3 for minor axis
>>> probe_node_3 = 3
>>> probe_angle_3 = "min" # here we insert the string "min" to get the minor axis values
>>> response.plot_magnitude(probe=[
...     (probe_node_1, probe_angle_1),
...     (probe_node_2, probe_angle_2),
...     (probe_node_3, probe_angle_3)
... ])
```
[image]

raphaeltimbo, 2020-12-02:
@LeifParr, I have opened the PR #713 so we can have plots with peak to peak results.
To use it, we just need to use the argument `amplitude_units` with the prefix `pkpk_`:
```python
response.plot([(0, 45)], amplitude_units='pkpk_m')

response.plot([(0, 45)], amplitude_units='pkpk_inch')
```

@leoishimoto, thanks for the feedback.

I am going to open two separate issues for that. 
One for:

> The major and minor axis of the ellipse is highly desirable not only the probes reference. In circular orbits does not make difference but for anisotropic bearings is necessary.

And another one for:
> For rolling element bearing’s the response in velocity / acceleration is also desired including the RMS values making the comparison more appropriate with field results.

LeifParr, 2020-12-03:
I thank you again for rapid response to my questions. 

One short comment to implementation of major/minor axis, that the keywords could be both "max" and "major" and "min" and "minor". Other rotordynamic software uses the terms "major" and "minor".

Another comment regarding the response plotting, is that the plots should indicate the node number (and the label) which are plotted. As far as I can see, there is no option to label probes, so there is room for confusion about what is being shown. If you consider this an important issue, you can move this into a new issue.

PR #713 (2020-12-03): Add prefix for peak to peak units
This adds the prefix `pkpk_` to our unit registry.
The user can now use this in plots as:

```python
import ross as rs
import numpy as np

rotor = rs.rotor_example()
speed = np.linspace(0, 1000, 101)
response = rotor.run_unbalance_response(node=3,
                                        unbalance_magnitude=.001,
                                        unbalance_phase=0.0,
                                        frequency=speed)

response.plot([(0, 45)], amplitude_units='pkpk_micron')
```
Since this is a prefix, we can use it with any length units: `pkpk_m`, `p [... cut]
closed 2020-12-03 by LeifParr