petrobras/ross #715: Add velocity / acceleration response plots

raphaeltimbo, 2020-12-02:
As mentioned by @leoishimoto in #710:

> For rolling element bearing’s the response in velocity / acceleration is also desired including the RMS values making the comparison more appropriate with field results.

rodrigomoliveira1, 2020-12-04:
@raphaeltimbo, I'm working on it.
I have a concearn about ao API with these new features. For example: in `FrequencyResponseResults` we have the follwing methods:
- `plot_magnitude`
- `plot_phase`
- `plot_polar_bode`
- `plot`
But, now, inserting velocity, acceleration, we'd need like 8 new functions, the 4 mentioned above to the velocity and 4 more to the accelaration.
- Plotting displacements:
    - `plot_displacement_magnitude`
    - `plot_displacement_phase`
    - `plot_displacement_polar_bode`
    - `plot_displacement`

- Plotting displacements:
    - `plot_velociy_magnitude`
    - `plot_velociy_phase`
    - `plot_velociy_polar_bode`
    - `plot_velociy`

- Plotting displacements:
    - `plot_accelaration_magnitude`
    - `plot_accelaration_phase`
    - `plot_accelaration_polar_bode`
    - `plot_accelaration`

Other option I thought would be adding an argument to the 4 original functions to pick one of the 3 options (displacement, velocity or accelaration). Maybe it would be a bit confusing to the user.

Any ideas?

raphaeltimbo, 2020-12-04:
I am thinking if changing the phase plots in this case would be useful, since we would just have a lag.
For now I was thinking of only changing the magnitude, and using the `amplitude_units` argument to achieve that.
Like, adding some code that would catch the exception when we tried to convert to 'mm/s' for example and in that part of the code maybe we could do the calculation.

I am also going to check other softwares to see how they treat that in their interfaces.

@leoishimoto

rodrigomoliveira1, 2020-12-04:
@raphaeltimbo, I followed your suggestion of manipulating `amplitude_units` arg. Look what I've done:
The code checks the unit passed to `amplitude_units`.
- If it is `'[length]/[force]'`, it displays the displacement;
- If it is `'[speed]/[force]'`, it displays the velocity;
- If it is `'[acceleration]/[force]'`, it displays the acceleration;

I had to create a `dummy_var` to use `pint.Quantity` syntax, but it's not a big deal.

```python
class FrequencyResponseResults(Results):
    def __init__(self, freq_resp, velc_resp, accl_resp, speed_range):
        self.freq_resp = freq_resp
        self.velc_resp = velc_resp
        self.accl_resp = accl_resp
        self.speed_range = speed_range

    def plot_magnitude(
        self,
        inp,
        out,
        frequency_units="rad/s",
        amplitude_units="m/N",
        fig=None,
        **mag_kwargs,
    ):
        """Plot frequency response (magnitude) using Plotly.

        This method plots the frequency response magnitude given an output and
        an input using Plotly.
        It is possible to plot displacement, velocity and accelaration responses,
        depending on the unit entered in 'amplitude_units'. If '[length]/[force]',
        it displays the displacement; If '[speed]/[force]', it displays the velocity;
        If '[acceleration]/[force]', it displays the acceleration.

        Parameters
        ----------
        inp : int
            Input.
        out : int
            Output.
        frequency_units : str, optional
            Units for the x axis.
            Default is "rad/s"
        amplitude_units : str, optional
            Units for the y axis.
            Acceptable units are:
                '[length]/[force]' - Displays the displacement;
                '[speed]/[force]' - Displays the velocity;
                '[acceleration]/[force]' - Displays the acceleration.
            Default is "m/N" 0 to peak.
            To use peak to peak use the prefix 'pkpk_' (e.g. pkpk_m/N)
        fig : Plotly graph_objects.Figure()
            The figure object with the plot.
        mag_kwargs : optional
            Additional key word arguments can be passed to change the plot layout only
            (e.g. width=1000, height=800, ...).
            *See Plotly Python Figure Reference for more information.

        Returns
        -------
        fig : Plotly graph_objects.Figure()
            The figure object with the plot.
        """
        frequency_range = Q_(self.speed_range, "rad/s").to(frequency_units).m

        dummy_var = Q_(1, amplitude_units)
        if dummy_var.check("[length]/[force]"):
            mag = np.abs(self.freq_resp)
            mag = Q_(mag, "m/N").to(amplitude_units).m
            y_label = "Displacement"
        elif dummy_var.check("[speed]/[force]"):
            mag = np.abs(self.velc_resp)
            mag = Q_(mag, "m/s/N").to(amplitude_units).m
            y_label = "Velocity"
        elif dummy_var.check("[acceleration]/[force]"):
            mag = np.abs(self.accl_resp)
            mag = Q_(mag, "m/s**2/N").to(amplitude_units).m
            y_label = "Acceleration"
        else:
            raise ValueError(
                "Not supported unit. Options are '[length]/[force]', '[speed]/[force]', '[acceleration]/[force]'"
            )

        if fig is None:
            fig = go.Figure()

        fig.add_trace(
            go.Scatter(
                x=frequency_range,
                y=mag[inp, out, :],
                mode="lines",
                line=dict(color=tableau_colors["blue"]),
                name="Amplitude",
                legendgroup="Amplitude",
                showlegend=False,
                hovertemplate=f"Frequency ({frequency_units}): %{{x:.2f}}<br>Amplitude ({amplitude_units}): %{{y:.2e}}",
            )
        )

        fig.update_xaxes(
            title_text=f"Frequency ({frequency_units})",
            range=[np.min(frequency_range), np.max(frequency_range)],
        )
        fig.update_yaxes(title_text=f"{y_label} ({amplitude_units})")
        fig.update_layout(**mag_kwargs)

        return fig
```

Examples of how it works:
```python
>>> import ross as rs
>>> rotor = rs.rotor_example()
>>> speed = np.linspace(0, 1000, 101)
>>> response = rotor.run_freq_response(speed_range=speed)

>>> response.plot_magnitude(inp=13, out=13, amplitude_units="m/N")
>>> response.plot_magnitude(inp=13, out=13, amplitude_units="m/s/N")
>>> response.plot_magnitude(inp=13, out=13, amplitude_units="m/s**2/N")
```

Outputs:

[image]

[image]

[image]

closed 2020-12-09 by raphaeltimbo