petrobras/ross #718: Include rotor speed on mode shape plots

LeifParr, 2020-12-08:
[image]

Considering how plotting of mode shapes is done in other rotordynamic program, here are some points which could be improved:
- Add possibility for custom title. For some of the other plots, I have added a title manually, but I'm not sure if this is straight-forward here, as the title field is used for vital info.
- Add the rotor speed. This is in my opinion essential, as the mode shapes may be plotted for several speeds, and we quickly loose track if the rotor speed isn't plotted per default.
- We use damping ratio, and I do not see a way to select damping ratio. 
- Contrary to the campbell diagram, I do not see a way to choose between Wn and Wd  for the natural frequency.

Anyway, this was only some suggestions, so far I do enjoy working with the ROSS module

rodrigomoliveira1, 2020-12-08:
Hello, @LeifParr, we're glad you're enjoying working wih ROSS so far. Thanks again for your suggestions, it's important to us to have this feedback.

[image]

So, answering your topics:
>Add possibility for custom title. For some of the other plots, I have added a title manually, but I'm not sure if this is straight-forward here, as the title field is used for vital info.

- We can add an argument to users input a breif title, more like a tag, as you can see in the figure above (ref 1). It's also possible to overlay the original title layout inserting the kwarg `title_text`, thus, you completely make your own title for the plot. This is, actually, a feature from [Plotly](https://plotly.com/python/reference/layout/) (link to the layout API reference). We only take advantage of this to allow the user to manipulate the `Plotly.Figure.layout` by inserting `kwargs`. Examples:

```python
>>> # Add fast title tag
>>> import ross as rs
>>> rotor = rs.rotor_example()
>>> modal = rotor.run_modal(speed=200)
>>> modal.plot_mode_3d(0, title="Rotor Model")
```
[image]

```python
>>> # Modifying title template 
>>> import ross as rs
>>> rotor = rs.rotor_example()
>>> modal = rotor.run_modal(speed=200)
>>> fig = modal.plot_mode_3d(1, title_text="NEW TITLE HERE")
```
[image]

> Add the rotor speed. This is in my opinion essential, as the mode shapes may be plotted for several speeds, and we quickly loose track if the rotor speed isn't plotted per default.

- Added as seen in ref. 2 (first figure).

> We use damping ratio, and I do not see a way to select damping ratio.

- Added as seen in ref. 4 (first figure).

> Contrary to the campbell diagram, I do not see a way to choose between Wn and Wd for the natural frequency.

- We're adding an argument (`frequency_type` - same argument that `run_campbell` has) to `plot_mode_3d` and `plot_mode_3d` to select what frequency to display. You can choose between "wd" and "wn". Example:

```python
>>> import ross as rs
>>> rotor = rs.rotor_example()
>>> modal = rotor.run_modal(speed=200, sparse=True)
>>> # Display damped natural frequency value
>>> modal.plot_mode_3d(1, frequency_type="wd", title="Rotor Model")
>>> # Display undamped natural frequency value
>>> modal.plot_mode_3d(1, frequency_type="wn", title="Rotor Model")
```

LeifParr, 2020-12-08:
Excellent. I trust you will also implement this for the 2d mode shape plots.

closed 2020-12-08 by LeifParr