petrobras/ross #721: Response results - DOF error when using n_link / point mass elements

LeifParr, 2020-12-09:
I have a model where I use casings for three bearings, using mass_elements and the n_link property for the bearings. My model has 84 shaft elements, so nodes 0-84 is shaft, node 85, 86 and 87 are housing elements. 
All of this works out nicely for the natural frequency calculations.

I move into unbalance response calculations, and my objective is to compare both relative shaft vibration (shaft node - housing node) and housing vibration (housing node) to measured vibration (proximity probes and housing accelerometers).

I tried extracting results for the housing-nodes, but ran into what I believe is a DOF-inconsistency.

I had to work through a number of these errors, in several functions in the results.py, example below:
  File "C:\Program Files\Python38\lib\site-packages\ross\results.py", line 1539, in plot_polar_bode
    phase = Q_(self.phase[p[0] * number_dof], "rad").to(phase_units).m
IndexError: index 348 is out of bounds for axis 0 with size 346

My temporary work-around (inserted into at least 3 functions in results.py):

Located in:
def plot_magnitude( 

etc. etc.

        shaft_elements = len(self.rotor.shaft_elements)+1 # lpa 2020-12-09
        if fig is None:
            fig = go.Figure()
# lpa 2020-12-09
        for i, p in enumerate(probe):
            if p[0] <= shaft_elements:
                dofx = p[0] * number_dof
                dofy = p[0] * number_dof + 1
            else:
                dofx = shaft_elements * number_dof + (p[0]-shaft_elements)*2
                dofy = dofx + 1

As far as I can understand, the mass_elements have only 2 degrees of freedom, so the indices can't be found simply by multiplying with number_dof (which is 4).

As an enhancement, I would very much like to have methods for extracting response magnitude / phase for a probe location etc as numpy arrays directly, without plotting them. As it is now, when I have to subtract to get relative vibration, I have to go into the plotly figure object and retrieve the series-data, which is a bit cumbersome.

rodrigomoliveira1, 2020-12-09:
Thanks for reporting that, @LeifParr. And thanks for your suggestion.

This kind of issue needs a wider view, because it certainly affects other functions along the code.
The `Rotor` object has an attribute called `nodes` that holds ONLY the nodes related to the shaft itself. I'm thinking about adding an attribute the holds the `n_link` nodes, which can be useful for other functionalities. So, in this specific case, we could do something like you did, but a bit simpler:

```python
    def _calculate_major_axis_per_node(self, node, angle, amplitude_units="m"):
        # collect necessary data
        ndof = self.rotor.number_dof
        nodes = self.rotor.nodes
        link_nodes = self.rotor.link_nodes

        # calculate DoFs
        fix_dof = (node - nodes[-1] - 1) * ndof // 2 if node in link_nodes else 0
        dofx = ndof * node - fix_dof
        dofy = ndof * node + 1 - fix_dof
```

LeifParr, 2020-12-09:
Thanks. I'll close this issue from my side.

I've run a fairly complex model, long shaft, three bearings with full set of stiffness, damping coefficients for every 100 rpm, and housing nodes with stiffness and mass. The plot below is a comparison of results from Ross and RAPPID. I think this is a fairly good match, considering that RAPPID is a little different when it comes to housing-modelling.

[image]

closed 2020-12-09 by LeifParr

rodrigomoliveira1, 2020-12-09:
> As an enhancement, I would very much like to have methods for extracting response magnitude / phase for a probe location etc as numpy arrays directly, without plotting them. As it is now, when I [...]

I will see what I can do about it. I'm expecting to create something general, that covers most of our `Results` class, to avoid repeating code lines.
And about the output structure, returning a `pandas.DataFrame` containing the results, seems to be a friendly strategy to the user, since you could better manipulate the output and easily export to a ".xls" or ".csv" file