petrobras/ross #726: Extract post-processed responses directly without plotting

rodrigomoliveira1, 2020-12-10:
It was suggested to us to create methods for extracting response magnitude / phase for a probe location, etc directly, without plotting them. As it is now, we have to go into the plotly figure object and retrieve the series-data.

GabrielBachmann, 2020-12-10:
For now, we are looking at the following functions:
- [ ] Campbell

- [ ] Time Response

- [ ] Frequency Response

- [ ]  Unbalance Response

rodrigomoliveira1, 2020-12-17:
Hello @LeifParr,

We're currently implementing a function that you can extract the data from a Plotly figure and store in a `pandas.DataFrame` format.
It has some limitations, for example: it supports only Scatter plots (X-Y), does not support subplots (multiple figures) and only 3 analyzes methods:
- `run_freq_response()`
- `run_unbalance_response()`
- `run_time_response()`

Example
```python
>>> import ross as rs
>>> rotor = rs.rotor_example()

# Run unbalance response
>>> resp = rotor.run_unbalance_response(
...     3, 0.001, 0.0, np.linspace(0, 1000, 101)
... )
>>> fig = resp.plot_magnitude(probe=[(3, 0.0, "probe1"), (3, np.pi/2, "probe2")])

# Get post-processed data from an unbalance response
>>> df = rs.get_data_from_figure(fig)

# Use the probe tag to navigate through pandas data
>>> df["probe1"]
Frequency (rad/s)    [0.0, 10.0, 20.0, 30.0, 40.0, 50.0, 60.0, 70.0...
Amplitude (m)        [0.0, 1.6057020543675704e-07, 6.63435909084514...
Name: probe1, dtype: object

>>> df["probe2"]
Frequency (rad/s)    [0.0, 10.0, 20.0, 30.0, 40.0, 50.0, 60.0, 70.0...
Amplitude (m)        [0.0, 1.733900136554049e-07, 7.18840693813769e...
Name: probe2, dtype: object

>>> df["probe1"][0]  # 0 - index for frequency array
array([   0.,   10.,   20.,   30.,...

>>> df["probe1"][1]  # 1 - index for amplitude array
array([0.00000000e+00, 1.60570205e-07, 6.63435909e-07, 1.57960208e-06,...

# Or use "iloc" to obtain the desired array from pandas
# df.iloc[response_type, probe]
>>> df.iloc[1, 0] # doctest: +ELLIPSIS
array([0.00000000e+00, 1.60570205e-07, 6.63435909e-07,...
```

I hope it helps you out.

rodrigomoliveira1, 2020-12-17:
About the `run_campbell`, it's way more complitcated to extract the data from the figure. The Scatters are plotted according to their `whirl direction`, so, getting the array from a trace would not help at all.

If you need to retrieve any natural frequency curve, I recommend you doing something simpler:
```python 

>>> rotor = rotor_example()
>>> speed = np.linspace(0, 400, 101)
>>> camp = rotor.run_campbell(speed)

# Each column from wd represents a natural frequency curve
>>> camp.wd.shape
(101, 6)

# Getting the first curve:
>>> camp.wd[:, 0]
array([91.79655318, 91.7965375 , 91.79649047, 91.79641208, 91.79630234,
       91.79616126, 91.79598883, 91.79578507, 91.79554997, 91.79528354,
       91.7949858 , 91.79465675, 91.79429641, 91.79390477, 91.79348186,...

# Getting the second curve:
>>> camp.wd[:, 1]
array([96.28899977, 96.28901106, 96.28904493, 96.28910137, 96.28918038,
       96.28928197, 96.28940612, 96.28955284, 96.2897221 , 96.28991392,
       96.29012828, 96.29036516, 96.29062457, 96.29090649, 96.29121091,...

# Getting the log dec curve:
>>> camp.log_dec[:, 0]
array([-2.54972794e-13, -3.95466693e-15, -3.97718081e-14, -9.85183268e-14,
       -6.17214825e-14,  3.64029636e-14,  4.67707806e-14, -1.10581127e-14,
        1.46075244e-14, -1.30286413e-14, -2.27002521e-13,  3.85211525e-14,

```

LeifParr, 2020-12-18:
Good afternoon. It may be that I didn't make myself clear in the original post. Extracting data series from figures is what I ended up doing (since the figures are returned, I could use my own code without the need for modifications to the ROSS library). But I did find that a bit cumbersome. Also, I noticed that the code for plotting of freq-response, time-response and unbalance in results.py seemed to have a segment which was shared, with code repeated. 

So my thought was to take that code and put it in a separate function inside of results.py. Something like:
Take response-results, or time results, at specified node and angle (or major/minor), and return amplitudes/phases for all calculated speeds. 

It seems kind of backward to me that we first put the desired data into a plotly figure, then extract the data from it afterwords.

closed 2020-12-23 by raphaeltimbo