petrobras/ross #728: Speed range and whirl values in run_critical_speed()

LeifParr, 2020-12-11:
I'm running a number of different calculation cases, and want to compare the critical speeds between them. I would like to try to sort the speeds automatically - for this purpose I need the whirl directions. I have put in a temporary work-around to this end (added calculation of modal.whirl_values() in run_critical_speed(), and added a parameter for the whirl values in the CriticalSpeedResults.

Could this be implemented in the major version? Alas, I do not yet have the pythonic level to contribute directly to the code.

As a side-note, I find the run_critical_speed routine a bit lacking. Typically, we do not calculate the bearings down to 0 rpm, in the Campbell below I have calculated the bearings from 100 to 900 rpm. This means that what happens below 100 rpm and above 900 rpm may be considered extrapolation / garbage, but I do not a see a way to limit the run_critical_speed-calculation to a speed range.

This is the second wish, a possibility for limiting the speed range of the run_critical_speed routine.

[image]

LeifParr, 2020-12-12:
As a little follow-up on the previous post, I have a temporary work-around. The idea was that I like the campbell diagram, so I want to extract the data from there. So what I did was to run through all of the data series in the campbell diagram, and find the ones that were labeled 'Crit. Speed'. Then, I would throw away empty series (there were a lot of them). For series with valid data, I take the x-value (or y-value for 1xcritical), and run_modal(x-value). I then check +-0.5% to find the one mode which is critical in this point, and extract the damping and whirl values. For some reason, there are duplicates in the plot, so I will have to remove these in the end. Cumbersome, but it works.

    for series in campbell_figure['data']: #Iterate through all series in Campbell diagram
       name = series['name']
        if name == 'Crit. Speed':
            print(name)
            speeds = series['x']
            if isinstance(speeds, np.ndarray):  # Check if speed is numpy array
                if speeds.size>0:               # Array has data?
                    speed = speeds[0]*np.pi/30.0 # convert back to radians for modal (campbell is in rpm)
                    mod_res = rotor.run_modal(speed) # Run modal at the critical speed to get damping and whirl values
                    for j, wd in enumerate(mod_res.wd): # Iterate through all modes in modal-results to find critical
                        if wd > 0.995 * speed and wd < 1.005 * speed: #Look for critical speed at +-0.5% of speed
                            crit_results.append([wd*30.0/np.pi, mod_res.damping_ratio[j], mod_res.whirl_values()[j]])

rodrigomoliveira1, 2020-12-14:
Hello @LeifParr, thanks again for your suggestions!

I'll take a look on this issue. I still have some other issues to take care, but this one is on the list.

I could change the optimization routine in `run_critical_speed()` to accept bounds, constraining the search range according to a `speed_range` argument the user enters. Furthemore, I could take the functions that calculate the whirl directions and add to `CriticalSpeedResults`. I will also allow users to change the frequency units output.

closed 2020-12-15 by raphaeltimbo