petrobras/ross #733: Plots not appearing

chadr03, 2020-12-23:
I am new to ROSS and trying to work through some examples.  None of the rotor plots are working.  I have tried using the example code in a Jupyter notebook as well as a stand alone text editor.  I do not get any errors but do not see any plots.  The Jupyter notebook appears to have white space for a plot, but no plot appears.  Any idea what my issue might be.  The following is the output I see when I try to run the provided example 3 notebook.

[image]

rodrigomoliveira1, 2020-12-23:
Hello @chadr03, thanks for your feeback.

The figures should appear automatically when calling a method that returns Plotly `go.Figure()` object

Since it's not working automatically, I recommend you adding the command `.show()` at the end. It has an optional argument called `renderer`, which you pick where you would like to display the figure.

Here is the [Plotly Displaying Figures documentation](https://plotly.com/python/renderers/) to help you with it.

Example 1 - In general it should work, if there's no pre-set default options
```python
rotor591c.plot_rotor().show()
```

If Example 1 does not work yet
Example 2 - If you want it to be displayed at Jupyter Notebook
```python
rotor591c.plot_rotor().show(renderer="notebook")
```

Example 3 - If you want it to be displayed at the browser
```python
rotor591c.plot_rotor().show(renderer="browser")
```

chadr03, 2020-12-23:
Thank you the .show() method seems to be working.

closed 2020-12-23 by chadr03