petrobras/ross #738: Travis has stopped working

rodrigomoliveira1, 2021-01-13:
If you have noticed in recent PR, TRAVIS CI hasn't been testing anything we have uploaded.
Then, I've been looking for the source of this problem, and look what is happening:

[image]

You can't see it right, but in the last column, it says that our repository does not have enough credits. 
Doing some quick research, I just found out that Travis CI announced new pricing model, effectively ending open source offering, and Travis-ci.org will be put into read-only mode.

Actually, there's still and free option to use TRAVIS, but we need to migrate from travis-ci.org to travis-ci.com.

Here is a [guide](https://docs.travis-ci.com/user/migrate/open-source-repository-migration/#frequently-asked-questions) to help us fixing this issue.

rodrigomoliveira1, 2021-01-13:
When migrating to travis-ci.com, we need to assign to one of their "plans". The free one limits the number of builds you can run, until you need to purchase more credits to keep using travis. 

[image]

[image]

raphaeltimbo, 2021-01-13:
@rodrigomoliveira1, as we were discussing, I think the option you were talking about with GitHub Actions is one that we should really evaluate. 
I have also seem a lot of projects migrating there, and we would have more things in one place only (github).
Also, I think github is more 'stable' in the sense that we probably won't have to this kind of migration again in a near future.

rodrigomoliveira1, 2021-01-22:
@raphaeltimbo, I'm doing a first try with GitHub Actions. I've just commited this [new file](https://github.com/ross-rotordynamics/ross/blob/master/.github/workflows/test.yml) to run our tests.
I'm starting with the simple, only adding pytest. We can also add codecov, black and Isort by different workflow files. It doesn't need to be all in one.

raphaeltimbo, 2021-01-27:
I am just reopening this for a last discussion about appveyor.
Since github actions are also testing windows builds, maybe we could remove the appveyor check?
I do not see any advantages using appveyor now. What do you think @rodrigomoliveira1 ?
If this is the case I will do a PR removing appveyor and also some files related to appveyor and travis ci.

rodrigomoliveira1, 2021-01-27:
I fully agree.

closed 2021-02-02 by raphaeltimbo