petrobras/ross #759: Some tutorials/examples are not showing cells outputs

raphaeltimbo, 2021-02-18:
As an example:
https://ross-rotordynamics.github.io/ross-website/examples/tutorial_part_2.html
[image]

rodrigomoliveira1, 2021-02-18:
At first, I thought it was caused due the lack of `.show()` command in the end of the line. But if we look to other examples from [tutorial part 1](https://ross-rotordynamics.github.io/ross-website/examples/tutorial_part_1.html), it shows the image without it.

[image]

raphaeltimbo, 2021-02-18:
For some reason notebooks will not always store the plots, so we have to execute them during the build process.
I have opened a PR implementing this.

raphaeltimbo, 2021-06-11:
I am reopening this issue since the plots are not showing again.

raphaeltimbo, 2021-06-17:
After migrating things to readthedocs I was still having this problem in some notebooks.
Just to register here, in one case I had to import plotly:
```python
import plotly.graph_objects as go
```
Even though I was not calling plotly directly in the notebook (was only the `rotor.plot_rotor()` method).

raphaeltimbo, 2023-05-26:
Once again we have this problem in the Modeling tutorial.
Will check what is happening.

raphaeltimbo, 2023-07-05:
Solved the current issue by trusting the notebooks, adding the following to the imports and rerunning:

```python
import plotly.graph_objects as go
import plotly.io as pio

pio.renderers.default = "notebook"
```

raphaeltimbo, 2024-03-22:
Reopen this on for the pages below:

- [ ] Overview
- [ ]

ViniciusTxc3, 2024-04-04:
Waiting for #1040 to be solved for this issue.

commit a53570dc (2021-02-18): Set nbsphinx_execute to always

This will run all notebooks during the docs building process. It avoids
having notebooks that have non working code in our documentation. It
also solves the problem of getting some notebooks without the cell
output (closes #759).
commit 8d0baa4c (2021-06-13): Add html_js_files for plotly

We need to add the html_js_files so that the plotly figures show up in
the docs.

Closes #759.
closed 2024-05-10 by ViniciusTxc3