petrobras/ross #772: Error when trying to run_ucs with 6dof rotor

raphaeltimbo, 2021-02-24:
When trying to run the `.run_ucs` method to a 6dof rotor I get the following error:

```python
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
<ipython-input-39-59650c74aad8> in <module>
----> 1 ucs0 = rotor.run_ucs()

~/ross/ross/rotor_assembly.py in run_ucs(self, stiffness_range, num_modes, num, fig, synchronous, **kwargs)
   2159         for i, k in enumerate(stiffness_log):
   2160             bearings = [BearingElement(b.n, kxx=k, cxx=0) for b in bearings_elements]
-> 2161             rotor = self.__class__(self.shaft_elements, self.disk_elements, bearings)
   2162             speed = 0
   2163             if synchronous:

~/ross/ross/rotor_assembly.py in __init__(self, shaft_elements, disk_elements, bearing_elements, point_mass_elements, min_w, max_w, rated_w, tag)
    198             raise ValueError("Tags should be unique.")
    199 
--> 200         self.number_dof = self._check_number_dof()
    201 
    202         ####################################################

~/ross/ross/rotor_assembly.py in _check_number_dof(self)
    530             len(brg.dof_mapping()) != number_dof / 2 for brg in self.bearing_elements
    531         ):
--> 532             raise Exception(
    533                 "The number of degrees o freedom of all elements must be the same! There are BEARING elements with discrepant DoFs."
    534             )

Exception: The number of degrees o freedom of all elements must be the same! There are BEARING elements with discrepant DoFs.
```

rodrigomoliveira1, 2021-02-25:
The problem is happening on line 2160.
```python
   2159         for i, k in enumerate(stiffness_log):
   2160             bearings = [BearingElement(b.n, kxx=k, cxx=0) for b in bearings_elements]
```
The code should select between `BearingElement` and `BearingElement6DoF` automatically.

raphaeltimbo, 2021-02-25:
Yes, I have replaced that with `b.__class__(...)`, but now I am getting other errors.
The problem now is that, for a 4dof we basically had to select the forward modes, but now we also have modes that are related to the torsional/axial dofs, and I still need to find a way to identify and remove those from the results. A hack could be to just create a 4dof rotor within the function using the 6dof rotor attributes.
But this problem made me think about how we are displaying our modes after running `.run_modal(...).plot(...)`, how can we show the user that a specific mode is related to torsion?

@rodrigomoliveira1 , @atbrandao

leiteamanda, 2022-11-24:
The solver's algorithms were evaluated to better understand the solution of the modes shapes, so that the possible solutions listed as viable options can be implemented in the already developed solver. The torsional and axial mode shapes were evaluated in order to represent them visually. To this end, the plot functions already implemented were examined to understand how to graphically represent the results of interest. The next step is to sketch the graphical solution.

closed 2024-08-22 by raphaeltimbo