petrobras/ross #773: Enhance the BallBearingElement Class

64J0, 2021-02-24:
Hi guys, can you please improve the BallBearingElement class in the docs?

I've read the docs and it only shows the parameters without an explanation on their meaning and a real-life example. Actually I'm trying to create a model to this bearing: https://www.skf.com/br/products/rolling-bearings/ball-bearings/deep-groove-ball-bearings/productid-6201-2Z.

Would you please help me create this model?

64J0, 2021-02-25:
I'm facing the same problem in XLRotor. It's documentation doesn't truly explains the parameters necessary to create a model of a ball bearing. Some illustrations would be helpful in this case haha.

64J0, 2021-02-25:
Other things that I've noticed. This examples are showing errors: 

* https://ross-rotordynamics.github.io/ross-website/examples/example_1.html
* https://ross-rotordynamics.github.io/ross-website/examples/example_2.html

64J0, 2021-02-25:
This code is giving me some errors. Can you explain why?

Code:

```python
N_bearing = 2

n_balls   = 7
d_balls   = 0.006
fs        = 500.0
alpha     = (np.pi * 8) / (6 * n_balls)
bearings  = [
  rs.BallBearingElement(
    n=i*2,
    n_balls=n_balls,
    d_balls=d_balls,
    fs=fs,
    alpha=alpha,
    tag=(f"Ball Bearing {i}")
  )
  for i in range(N_bearing)
]
```

Error:
```python
ValueError                                Traceback (most recent call last)
<ipython-input-5-7a53838ec39e> in <module>
      8 fs        = 500.0
      9 alpha     = (np.pi * 8) / (6 * n_balls)
---> 10 bearings  = [
     11   rs.BallBearingElement(
     12     n=i*2,

<ipython-input-5-7a53838ec39e> in <listcomp>(.0)
      9 alpha     = (np.pi * 8) / (6 * n_balls)
     10 bearings  = [
---> 11   rs.BallBearingElement(
     12     n=i*2,
     13     n_balls=n_balls,

~/.local/lib/python3.8/site-packages/ross/bearing_seal_element.py in __init__(self, n, n_balls, d_balls, fs, alpha, cxx, cyy, tag, n_link, scale_factor)
   1237         else:
   1238             f = interpolate.interp1d(nb, ratio, kind="quadratic")
-> 1239             kxx = f(n_balls)
   1240 
   1241         if cxx is None:

~/.local/lib/python3.8/site-packages/scipy/interpolate/polyint.py in __call__(self, x)
     76         """
     77         x, x_shape = self._prepare_x(x)
---> 78         y = self._evaluate(x)
     79         return self._finish_y(y, x_shape)
     80 

~/.local/lib/python3.8/site-packages/scipy/interpolate/interpolate.py in _evaluate(self, x_new)
    675         y_new = self._call(self, x_new)
    676         if not self._extrapolate:
--> 677             below_bounds, above_bounds = self._check_bounds(x_new)
    678             if len(y_new) > 0:
    679                 # Note fill_value must be broadcast up to the proper size

~/.local/lib/python3.8/site-packages/scipy/interpolate/interpolate.py in _check_bounds(self, x_new)
    704         # !! Could provide more information about which values are out of bounds
    705         if self.bounds_error and below_bounds.any():
--> 706             raise ValueError("A value in x_new is below the interpolation "
    707                              "range.")
    708         if self.bounds_error and above_bounds.any():

ValueError: A value in x_new is below the interpolation range.
```

rodrigomoliveira1, 2021-02-25:
Hello, @64J0! Thanks for your feedback.

I will work on a better docstring for `BallBearingElement`, so parameters can be a little bit clearer.
The fact is, several references uses the contact angle between spheres and inner/outer raceways and the loading as input parameters to calculate the bearing stiffness. I understand it's not much practical, but I haven't found a reference that uses the same parameters set you can find on a catalogue.

About the examples, something is odd because it says it can not find matplotlib library.

And about the last error, i will change the interpolation command so it can also extrapolate values.

raphaeltimbo, 2021-02-25:
Regarding some of the error examples, matplotlib has been removed from our requirements after we moved our plots to plotly.
I will try to update this examples to use plotly instead of matplotlib.

64J0, 2021-02-25:
Thanks @rodrigomoliveira1, I think you're right, XLRotor's approach use those terms that you mentioned too and I can't figure out how to get them with the datasheet x.x. Some time ago I've found this software called Mesys that can calculate the bearing props but doesn't looked very deep.

64J0, 2021-02-25:
Thanks @raphaeltimbo, I'll update my ROSS local version too.

rodrigomoliveira1, 2021-02-25:
@64J0,

I just created a PR adding the reference we're using to calculate the coefficients. You may check it out to get a closer look on the model.
I have also fixed the interpolation issue by adding an argument that allows extrapolations.

64J0, 2021-02-25:
Awesome @rodrigomoliveira1 , thanks. I'll close this issue now.

closed 2021-02-25 by 64J0