petrobras/ross #780: Improvement on method .plot_rotor()

CisneirosRaphael, 2021-03-01:
When you use .plot_rotor the model displays same size disks even the values mass/inertia/diameters being different.
It makes validation a bit harder if the user gives any wrong input.
My recommendation is to plot proportionaly different disks sizes with regard to mass/inertia or diameters.

rodrigomoliveira1, 2021-03-01:
Hello @CisneirosRaphael. Thank you for your feedback!

While we do not have such feature, I'd recommed you to use `scale_factor` argument to scale the glyph size. Check [DiskElement API](https://ross-rotordynamics.github.io/ross-website/generated/elements/ross.DiskElement.html#ross.DiskElement). 
Setting `scale_factor` smaller than 1, reduces the glyph size, and setting it greater than 1, inreases its size. You can set each `scale_factor` proportionaly to their masses/inertia manually.

raphaeltimbo, 2021-05-25:
Hi @CisneirosRaphael !
I have changed the code to allow the following when creating disks:
```python
disk = DiskElement(n=0, m=32, Id=0.2, Ip=0.3, scale_factor='mass')
```
If you create a rotor were all disk elements hava `scale_factor='mass'`, the code will calculate automatically calculate a scale factor based on the disk mass.

closed 2021-05-25 by rodrigomoliveira1

raphaeltimbo, 2022-06-14:
> Hello everyone, I am new working with this tool, but I would like to ask a question: when I use .plot_rotor() command, what I obtain is a squeezed text of hundred of lines, insted of a plot. I do have available plotly lib, but I do not know exactly what is happening. I would really appreciate if someone could help me with this issue. Thank you in advance!

Hi @Maite2707.
I can look at your problem.
Please, if you could open a new issue and provide code to reproduce your problem and also a figure from the plot you are getting.

Maite2707, 2022-06-15:
Hi Raphael,

Thank you for your answer. I opened a new issue:
https://github.com/ross-rotordynamics/ross/issues/891

Thank you for your support!

PR #796 (2021-05-25): add scale factor mass
- Add option for scale_factor='mass' in disks

This adds the option to use scale_factor='mass' when creating the disks.
If all the disks given to the rotor constructor have this scale factor,
the value for the scale factor will be calculated when the rotor is
instantiated based on the disk with the highest mass.
The disks with mass close to the highest value will have scale_factor=1
and disks with lower masses will have scale factors that will be closer
to 0.5.

Closes #780.