petrobras/ross #783: CampbellResults.plot() - damping ratio and filtering

LeifParr, 2021-03-11:
I have started using Ross 1.0.0, and have some two suggestions for improving plotting of Campbell results. My goal is to incorporate changes that may be of interest to other users, so I do not have to change the modules manually every time I update.

Damping ratio: We use damping ratio, not log dec. There should be an option for plotting using damping ratio:
1. The damping ratio could be passed to CampbellResults, or calculated in plot. damping = 1.0/np.sqrt(1+(2*np.pi/self.log_dec)**2)
2. The color axis should be reversed when using damping ratio: coloraxis_colorscale="rdbu"
3. Change text: coloraxis_colorbar=dict(title=dict(text="<b>Damping ratio</b>", side="right"))

Filtering: The RAPPID rotordynamic software has options for filtering the modes shown on the campbell diagram. For example, modes with high damping may not be of interest, so we can supply options for a range of damping ratios that should be included (or left out) of the plot. Also, I had some 0.5 cpm modes that I didn't want to show, so filtering of frequencies could also be optional. Filtering of machine speeds is easily obtained by limiting the X-axis, which may be done outside of the plot function. 
I did this filtering manually, something like:
                whirl_mask = whirl_i == whirl_dir
                # lpa 2021-02-25 added a second masking for high damping, and zero modes
                damping_mask = log_dec_i < lpa_damping
                freq_mask = w_i > lpa_low_freq
                lpa_mask = whirl_mask & damping_mask & freq_mask

raphaeltimbo, 2021-03-11:
Hi there, @LeifParr !
These are excellent suggestions and I would like to have them incorporated to the ROSS code.
From what I understand you already have the code for that. Would you like to make a PR?

LeifParr, 2021-03-11:
For now, I don't have Github setup properly to make pull requests and so forth, and I consider my level of programming a bit too low for contributing actual code. So I would prefer if someone else could do the actual PR...

raphaeltimbo, 2021-03-11:
No problem. I will implement that and will let you know about the progress.

raphaeltimbo, 2021-04-12:
@LeifParr , I have started implementing this and wanted to know your opinion regarding the filter for frequency and damping.
I am thinking about adding the arguments `frequency_range` and `damping_range`.
Both would be a tuple with the format `(min, max)`. What do you think?
For choosing between log_dec and damping ratio I was thinking about an argument called 'damping' (still in doubt with we could have a better name here) and the user would provide a string 'log_dec' or 'damping_ratio'.
Regarding the coloraxis, I did not understand why we should reverse it, since I consider a high log dec and a high damping ratio a good ('blue') thing.

LeifParr, 2021-04-13:
_frequency_range_ and _damping_range_ sounds fine to me. None will be default, which will be no filtering?
For choosing between log_dec and damping ratio:
_damping_parameter_ is a possibility, so is _damping_type_. Other rotordynamic software uses the term _stability_parameter_, however, this may possibly be misinterpreted, so I vote for _damping_parameter_.

Regarding color axis, you are absolutely right. It was a glitch from my side, I saw that I hadn't actually changed that in my own 0.4.1 version, it just passed through as a possibility.

raphaeltimbo, 2021-04-13:
> _frequency_range_ and _damping_range_ sounds fine to me. None will be default, which will be no filtering?
Yes, None will be default with no filtering.

> _damping_parameter_ is a possibility, so is _damping_type_. Other rotordynamic software uses the term _stability_parameter_, however, this may possibly be misinterpreted, so I vote for _damping_parameter_.

Agreed on damping_parameter!

closed 2021-04-14 by raphaeltimbo

raphaeltimbo, 2021-04-14:
@LeifParr , I have just merged the code and also released a minor ROSS version (v1.0.1) with all the recent modifications.
Let me know if you have any problems with the code.

PR #788 (2021-04-14): Add damping parameter and filtering to campbell
This PR adds three additional arguments to the Campbell plot:
- damping_parameter: choose between log_dec or damping_ratio;
- frequency_range: filter out modes that are out of this frequency range;
- damping_range: filter out modes that are out of this damping range;

Closes #783.